lgsvl/simulator #1525: Cluster configuration feature requests in web UI

RobShewagaUHN, 2021-06-28:
Hello, two feature requests for managing clusters in the web UI:

1. The ability to have one machine be in two different clusters. I want my dev machine to be in a cluster by itself for solo testing, and also be in a second cluster where other machines are attached to it. Currently the Add New button can only start creating a new cluster when you link a new PC to the cloud, instead of selecting an existing machine and starting from there.
2. The ability to manually add in a machine to a cluster without having to link that machine to the cloud (a fully offline machine). I think I remember this was the old way of doing it - manually adding a name, IPs, etc.

lemketron, 2021-06-28:
Hi @RobShewagaUHN,

One thing that comes to mind is using two simulator executables in different folders. You can launch them from the command line and pass (with `--data` or `-d`) a path to the local database (storage of assets as well as the unique ID used when linking to a cloud cluster).

Command line parameters are documented here: 
https://www.svlsimulator.com/docs/user-interface/config-and-cmd-line-params/#command-line-parameters

EricBoiseLGSVL, 2021-06-28:
adding @iamdavidfrank