lgsvl/simulator #1526: Lidar and Radar don't produce points at the same location for the same object.

MaxandreOgeret, 2021-06-29:
Hello, 
I am using ROS1 and I am trying to setup a configuration with one lidar and one radar.

The radar and the lidar are aligned (no rotation between the two)

Radar config : 
[image]

Lidar config : 
[image]

As you can see both are not rotated relative to base link.

In ROS the Tf is as is and has no rotation : 

```
rosrun tf tf_echo /lidar /radar_fc          
At time 0.000
- Translation: [2.980, -0.000, -1.717]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

```

I use the ros bridge and I display the radar points alongside the lidar points : 

[image]

As you can see the lidar and the radar are mismatching.

Here is how I extract the position of the radar objects to display them : 

```python
def tf_point(p, tf_listener, map_frame, radar_frame):
    point_stamped = PointStamped()
    point_stamped.header.frame_id = radar_frame
    point_stamped.point = Point(x=p.x, y=p.y, z=p.z)
    tfed_point = tf_listener.transformPoint(map_frame, point_stamped).point
    return Point(x=tfed_point.x, y=tfed_point.y, z=tfed_point.z)

tfed_point = Point()
tfed_point = radar_tf.tf_point(Point(), self.tf_listener, self.map_frame,
                               self.radar_frame)

marker.header.frame_id = self.map_frame
marker.pose.position.x = tfed_point.x + radar_obj.object_relative_position.x
marker.pose.position.y = tfed_point.y + radar_obj.object_relative_position.y
marker.pose.position.z = tfed_point.z + radar_obj.object_relative_position.z

```

---

Given that : 
- The TF between lidar and radar in the web interface has no rotation
- The TF between lidar and radar in ROS has no rotation
- The object points from the lidar and the radar are mismatching. (It seems that the lidar is fine, but the radar is rotated somehow)

It looks like a bug. 

What do you think ? 

Thanks !

EricBoiseLGSVL, 2021-07-07:
Odd that there is a slight rotation.  We have switched Lidar sensor to use a cubemap solution in the next release.  Let's see if this non rotating version has the same issue

MaxandreOgeret, 2021-07-07:
When the new version is out I will gladly test again.

bruce-almon, 2021-07-08:
Can I ask how did you get the radar to work with ros1?
I tried rostopic echo /radar_fc but nothing (not even header and frame ID) was printed even though the radar visualization (3D bounding box) in lgsvl is ok .
I am using lgsvl 2011.1.1

Thanks