lgsvl/simulator #1527: vehicle_status

mengen100, 2021-06-29:
Does the latest LGSVL support vehicle_status topic? 
I'm running LGSVL 2021.1 with autoware.ai 1.14, but I topic vehicle_status seems empty in autoware.ai.
Your kind help is much appreciated! Thanks!

EricBoiseLGSVL, 2021-06-29:
@hadiTab do we have plans to support this sensor or should users create their own?

mengen100, 2021-07-16:
@EricBoiseLGSVL Would it be easier if you could share how to create my own _vehicle_status_ for LGSVL?

EricBoiseLGSVL, 2021-07-16:
If you look at the existing code you can see that you need conversion methods in the bridge code and the data type to be collected in the sensor.  We still have some code left over in simulator for bridges that may need modified as well.  @hadiTab can let you know the best way.  I think an existing sensor that you add the data to would be best.