lgsvl/simulator #1528: Cannot switch accounts in one computer

krtt, 2021-06-29:
I want to run Apollo in one computer (computer A) and the SVL sim in another (computer B). Both are Linux 18.04 and both computers have been used to run a pre-built SVL binary but linked to separate SVL user accounts (computer A to SVL user cloud A and computer B to SVL user cloud B). Now, when I am a new user in computer B and run the SVL binary, it does not let me connect to my cloud A (button "Go online" or unlink from the cloud (button "Unlink"). These buttons don't do anything when I click them. The sim executable window just sits in "Offline" and reacts only to 'Clear Cache' and 'Quit'. 
So in the browser I cannot create a new cluster in cloud A using computer B. It just asks me to download the binary and link to the cloud. What is correct to do?

EricBoiseLGSVL, 2021-06-29:
@thisisdavidfrank @eagafonovlge ?

heeen, 2021-07-06:
as for multiple user profiles, shouldn't adding `-d someExistingDirectoryA` or `-d someDirB` allow switching between accounts