lgsvl/simulator #1530: Radar velocity is wrong

MaxandreOgeret, 2021-06-29:
Hello, 
I continue my tests with the radar and I think there is a discrepancy in the reported velocity.

Here is how I get and compute the norm of the velocity vector : 

```python
from numpy import linalg

class sim_radar_detector:
    def __init__(self):
        rospy.Subscriber('/radar_fc', DetectedRadarObjectArray, self.radar_obj_array_cb)

    def radar_obj_array_cb(self, radar_objects):
        i = 0
        for radar_obj in radar_objects.objects:  # type: DetectedRadarObject
            markers.markers += self.generate_marker(i, radar_obj)
            i += 1

    def generate_marker(self, marker_id, radar_obj):
        print("relative_velocity: [%d] %.2f" % (marker_id, linalg.norm([radar_obj.object_relative_velocity.x, radar_obj.object_relative_velocity.y, radar_obj.object_relative_velocity.z])))
        print("velocity:          [%d] %.2f" % (marker_id, linalg.norm([radar_obj.object_velocity.x, radar_obj.object_velocity.y, radar_obj.object_velocity.z])))
        print("----")

```

You can see that I calculate the norm of the velovity vectors `object_relative_velocity` and `object_velocity` and print them.

Here's the video of the output : 

https://user-images.githubusercontent.com/22786578/123832739-55f7ba00-d90e-11eb-8291-cfb6820bbcf1.mp4

You can see that both values are almost zero and give no valuable output.

I would expect those values to be : 

- **object_velocity: ** Object velocity relative to the radar frame, projected on the map frame vectors.
- **object_relative_velocity: ** Object velocity relative to the map frame, projected on the map frame vectors.

But what is happening here is clealy not that. Am I missing something ? 

Thanks !

EricBoiseLGSVL, 2021-09-13:
We will look into this.  I'm guessing you are moving vehicles with API? The velocity is calculated from the rigidbody.  Not sure why that value would be wrong.

EricBoiseLGSVL, 2021-09-16:
We have a fix for next release.  Thanks for posting to let us know about this.

closed 2021-09-16 by EricBoiseLGSVL