lgsvl/simulator #1531: Sensor distribution feature requests

RobShewagaUHN, 2021-06-29:
Hi all, here's some feature requests for sensor distribution methods.

1. The ability to designate a sensor to run on every machine in the cluster - not only one machine per sensor [as per documentation](https://www.svlsimulator.com/docs/distributed-simulation/sensors-distribution/). This would be useful in the case of, for example, having a sensor that loads some StreamingAssets resources useful to each independent machine in the cluster.

2. Explicit sensor delegation to target machines in the cluster. Currently, the master delegates sensors automatically to various clients [based on a load distribution algorithm](https://www.svlsimulator.com/docs/distributed-simulation/sensors-distribution/). It would be useful to be able to explicitly configure the sensors in a cluster to target machines as wanted.

EricBoiseLGSVL, 2021-06-29:
Thanks for the requests.  @zelenkovsky

EricBoiseLGSVL, 2021-06-29:
pinging @KarByc to track suggestion 2