lgsvl/simulator #1532: creating accident scenario for pedestrain

rajeshjoshi8792, 2021-06-30:
I want to create accident scenario for pedestrain. like vehicle moving fast in the road. and suddenly pedestrain walk in the road. and vehicle collision with pedestrain.

Right now i able to create this scenario. but when vehicle collision with pedestrain, vehicle is stopping or sometime pedestrain jump above vehicle. i want to create real accident scenario where pedestraion fall in the road when vehicle collision.

EricBoiseLGSVL, 2021-06-30:
That is not supported at this time.  You will need to create animations and or rag doll support for the pedestrian.  Then you will need to create trigger events on collision.  EGO will not brake if you are driving but an AD stack will.  How will you control the vehicle?

lemketron, 2021-07-01:
Note that SVL Simulator is not intended for properly simulating what happens AFTER a collision. The fact that a collision took place (contact between the ego and something like an NPC or pedestrian) should cause the simulation to fail and indicate (in the test report) that a collision occurred.

closed 2021-07-06 by lemketron