lgsvl/simulator #1533: signaldata.csv and vector.csv in the map of SanFrancisco is empty

windwolf9, 2021-06-30:
hello,
I'm testing the traffic light module，but i Can't get the effect,and I debug the program and found the reason is signaldata.csv and vector.csv in the map of SanFrancisco is empty. 
I downloaded it from : https://github.com/lgsvl/autoware-data .Can you provide the data?
The platform I use：autoware1.13 + lgsvl2020.06 (SanFrancisco)
Thank you very much！

EricBoiseLGSVL, 2021-06-30:
@windwolf9 If you are using 06 release, and following current release documentation you won't be able to use some features.  What documentation are you following?

windwolf9, 2021-07-01:
@EricBoiseLGSVL Thank you for your reply, What you replied may not be the cause of the problem. I run the trafficlight_recognizer in autoware1.13.0 , the node of  “feat_proj” need signaldata.csv and  vector.csv,but i open " /shared_dir/autoware-data/SanFrancisco/data/map/vector_map/signaldata.csv",it is empty. 
I run with  ” BorregasAve ”map, it is ok, I checked the BorregasAve map(/autoware-data/BorregasAve/data/map/vector_map/signaldata.csv), it is not empty.
Thank you very much！

EricBoiseLGSVL, 2021-07-02:
What version of simulator are you using?  In older versions of simulator, signaldata.csv will be empty for SF map and not Borregas. What version are you using?  Once we confirm this we can look into why this is missing for you.

windwolf9, 2021-07-06:
@EricBoiseLGSVL Thank you for your reply!  I use the simulator version is “lgsvlsimulator-linux64-2020.06”.
You are right ,autoware-data( https://github.com/lgsvl/autoware-data) is too old, signaldata.csv and vector.csv in the vector_map of SanFrancisco is empty, only  signaldata.csv and vector.csv in ” BorregasAve ”map is not empty.

EricBoiseLGSVL, 2021-07-27:
Autoware vector maps do not have signal data supported. I think this is an older format that we haven't added to in a while.  I'll ask internally on this. Please do not post duplicate issues.

windwolf9, 2021-07-27:
ok , thank you very much，If there is any new progress, would you please reply here, thanks!

EricBoiseLGSVL, 2021-07-27:
No problem.  Missing signaldata is expected for autware vector map.  Since Autoware.AI is near end of life, we put work into supporting Autoware.Auto and lanelet2 for this data.

closed 2021-09-14 by EricBoiseLGSVL