lgsvl/simulator #1535: How to run multiple simulation at a time in local cluster

tecena, 2021-06-30:
Hi,

we are using the executable file to launch the simulator on windows and have created  a local  cluster and running the simulation. With this setup we can run only one simulation and execute  the python script in API-only mode.If we try to start another simulation ,it throws error. Is there a way we can start multiple simulation parallelly on local cluster. Please  help us.

EricBoiseLGSVL, 2021-06-30:
Are you trying to run multiple simulations on one machine?

tecena, 2021-07-01:
> Are you trying to run multiple simulations on one machine?

Thank you for the reply @EricBoiseLGSVL
Yes, we tried on a single  machine,but we created two instances using simulator.exe --data <folder path> and the same cluster we linked to both these instances (using link to cloud button).Then executed the simulation using the python-API.But it did not work.
after this we tried using two windows machine and ran the simulator executable linking to same cluster, But that too did not work.
Please help us in figuring out how can we run simulations simultaneously using python-API

EricBoiseLGSVL, 2021-07-02:
We do not support one machine running multiple simulator instances.  You need to set a different data_path for each instance in the config.yml.