lgsvl/simulator #1536: SQLite exception

akshatamk, 2021-06-30:
Hi,

I am setting up the simulator with build from source on windows. I have downloaded unity hub, when I build the simulator in unity hub, the simulator executable file getting generated. I'm launching the .exe file by double clicking. The simulator opens with this error.
[image]

Next when I click link to cloud its not working.

Please help

EricBoiseLGSVL, 2021-06-30:
Go to your AppData folder for simulator and delete the entire folder
```
C:\Users\USER\AppData\LocalLow\LGElectronics\SVLSimulator
```
Then link the machine to the cluster.  Also, be sure you have space on the drive.

KeyWongQD, 2021-07-02:
I'm the same as you. Deleting folder didn't work.
I think the error might be in Docker ,but I don't know how to solve.

EricBoiseLGSVL, 2021-07-02:
Are you out of space on drive?  By deleting the library folder you should get a new link to cloud.  Is that the case?
@thisisdavidfrank can you think of any solutions for this?

lemketron, 2021-07-02:
Can you (quit and re-start the simulator and then) share the Player.log file?

heeen, 2021-07-06:
I believe this happened a while  ago when you deleted the sqlite db in the persistent data path or had an out of date sqlite db. Unpacking the simulator with a newer version and re-linking could fix it.