lgsvl/simulator #1537: Apollo stops detecting traffic lights when EGO reached a stop line

JuliaEmelianova, 2021-06-30:
**Description**
When EGO drives forward to the crossroad with traffic signals, Apollo detects traffic signals correctly. 

[image]

However, as soon as EGO reaches  a stop line, Apollo stops detecting traffic signals and "NO SIGNAL" is shown in Dreamview window.

[image]

[image]

I see this issue in both SanFrancisco and BorregasAve maps.
I think this issue could lead to incorrect EGO's behavior, e.g., when it is waiting a pedestrian crossing the road at the red light. In such a case, EGO continues its motion at any traffic signal, red, green or yellow.

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

ntutangyun, 2021-07-01:
Maybe you could try to take a look at the cameras images using cyber_visualizer.

EricBoiseLGSVL, 2021-07-01:
Is this Apollo 6.0/master? It is known that Apollo perception is bugged, that is why we use modular testing.  Have you followed up on the Apollo github issues for this?

JuliaEmelianova, 2021-07-01:
@EricBoiseLGSVL It is Apollo master from https://github.com/ApolloAuto/apollo

lemketron, 2021-07-01:
Are you using [Modular Testing](https://www.svlsimulator.com/docs/tutorials/modular-testing/) (simulator ground truth) or Traffic Signal perception in Apollo?

If you're just saying that the information in DreamView goes to "no signal" once the ego starts moving after the light turns green, this is normal behavior. Once the ego sees the signal turn green, it resumes driving and that signal information is no longer presented to (or relevant to) Apollo. So the status goes to "no signal" until the ego is within range of a new signal.

closed 2021-07-02 by EricBoiseLGSVL

JuliaEmelianova, 2021-07-04:
@lemketron I am using Modular Testing in Apollo (SVL Content for Lincoln2017MKZ is Apollo 6.0 (modular testing))