lgsvl/simulator #1538: Lane creation and road marking issues

honey5967, 2021-07-01:
Hi, I'm using SVL's OpenDRIVE function just fine..! Thank you..!
This time we have a problem with the road construction and we will contact you. If you have a shoulder road other than DRIVEING, the OpenDRIVEMapImporter script works fine, but when the script completes, no shoulder lane IDs are generated... and curb lane markings don't apply either.
Oh, I'm creating a road in play mode by declaring a static variable that reflects the width value of the OpenDRIVE lane.
This is the unfold code I changed.
[image]
[image]
[image]
Maybe there's a problem with the code I've extended?

EricBoiseLGSVL, 2021-07-01:
Sorry, I'm not clear on what the issue is.  Are you asking why your code changes are not working?  Is this post the extent of the changes?  Why is it in playmode?  What errors/issue are you seeing with the code changes you have?
Are you expecting shoulder lane IDs and curb lane markings? I don't think those are supported at this time.  Right @kamkolak-lge

honey5967, 2021-07-02:
hi eric thanks for replying
I contacted you because the changed code did not work as intended First of all the lane creation problem was because I commented out the CleanUp() function in Calculate(), so all the lanes appeared fine..!
The reason I did it in play mode was that the error occurred when I declared a global variable and tried ImportRoad more than once. But I solved it by clearing the global variable when creating the road..!
Now I can get all roads and lanes, but the lateralProfile of the road doesn't seem to apply. Are you planning to support lateralProfile in svl?
Thank you.