lgsvl/simulator #1539: Apollo does not detect some crosswalks

JuliaEmelianova, 2021-07-01:
**Description**
Not all crosswalks seen in SVL Simulator are detected by Apollo. E.g., Apollo does not detect a front crosswalk on SanFrancisco map:

[image]

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo master (latest Apollo was cloned at 2021-06-05)
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

ntutangyun, 2021-07-01:
the crosswalk might be missed during the HD map labeling process.

EricBoiseLGSVL, 2021-07-01:
All crosswalks are annotated but I think they don't have unique IDs so Apollo is having issues with it.  It's a bug that we have filed, thanks for the reminder.

JuliaEmelianova, 2021-07-01:
Thanks for your replies