lgsvl/simulator #1541: The problem of waypoint information in map

EthanCodesss, 2021-07-03:
Hello!
First of all, I encountered  a problem while developing. Let me talk about what i want to do:I want to use ros to make a simple AD stack to test my own motion planning algorithm. Then I first need to get the map topology information to do simple route planning. In apollo, they convert hd map to directed graph for route planning. But apollo is heavy for me. I only need a simple route for my planning module. So I can focus on the planning algorithm. 
My question is: how can i get the map information using python api? And is there a example to convert hd map and plan a route?
Second: I am unfamiliar with Unity.  Does the index inget_spawn()  function in pythonapi correspond to the index in the map annotation?
If anyone can help me, I would be very grateful!!

dylanyd, 2021-07-03:
In '<Apollo-path>/modules/tools/mapshow' folder, there is a python script 'mapshow.py', which can read HD maps and plot them. You can modify this script to get the map information.

EthanCodesss, 2021-07-03:
> In '/modules/tools/mapshow' folder, there is a python script 'mapshow.py', which can read HD maps and plot them. You can modify this script to get the map information.

I try it, but i got:
`/apollo/./bazel-bin/modules/tools/mapshow/mapshow.runfiles/apollo/modules/tools/mapshow/mapshow.py:119: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.show()
`

EricBoiseLGSVL, 2021-07-06:
@EthanCodesss Sorry, we do not provide map annotation data in the API.  You can create a sensor to read the data in simulator from MapManager.cs.  You could also edit the python api to parse this data as well but we don't have plans to do this in the near future.  Get spawn gets the Unity component SpawnInfo.  This sets the spawn position and has nothing to do with the index.

EthanCodesss, 2021-07-07:
> @EthanCodesss Sorry, we do not provide map annotation data in the API. You can create a sensor to read the data in simulator from MapManager.cs. You could also edit the python api to parse this data [...]

Thanks for your reply!! In the PythonAPI document, the npc has following waypoints function: npc.follow(waypoints, loop=True). And the waypoints is design manualy ( lgsvl.DriveWaypoint(.....) ).  For example, how can i get the waypoints information if i want the npc vehicle/pedestrian follow a specific path?  Is the vector position information viewd in unity?

EricBoiseLGSVL, 2021-07-07:
We recommend using our Visual Scenario Editor or VSE.  This only works with Linux at the moment though.  This allows you to place waypoints in the scene.  If you don't have linux you can still use it and then copy out the waypoints from the JSON VSE outputs or run the editor and mark points in the scene at run time.

EthanCodesss, 2021-07-07:
> We recommend using our Visual Scenario Editor or VSE. This only works with Linux at the moment though. This allows you to place waypoints in the scene. If you don't have linux you can still use it [...]

Thanks for your kindness and patience. I tried the VSE, what confusing me is: Is it only support the apollo cyber bridge? I'm using ros to develop the AD stack. In the simulation page, which type of vehicle should I choose?

EricBoiseLGSVL, 2021-07-07:
@EthanCodesss yes it only supports Apollo cyber at this time but we are working on custom VSE runners and it should release soon.

EthanCodesss, 2021-07-08:
> @EthanCodesss yes it only supports Apollo cyber at this time but we are working on custom VSE runners and it should release soon.

OK. Thanks!  The simulator is really amazing.  I get all I want to know.  Question will be closed.

closed 2021-07-08 by EthanCodesss