lgsvl/simulator #1543: Should EGO drive after a collision or not

JuliaEmelianova, 2021-07-04:
Could you clarify please what is an expected behavior of EGO after it collides with a pedestrian, NPC or some object? Should it continue its movement or should it remain be stopped?

lemketron, 2021-07-06:
Usually a collision indicates the end of a failed test and should reported as such in the [test results](https://www.svlsimulator.com/docs/user-interface/web/test-results/).

JuliaEmelianova, 2021-07-06:
@lemketron Thanks for the reply. If I understand correctly, it is not important at current stage of project, isn't it? Such a question arised due to the possible tests related to an expected behavior of EGO vehicle after a collision.

lemketron, 2021-07-06:
SVL Simulator is intended for testing of autonomous vehicles (and other robots) and not so much for accurate accident reconstruction (after the point of impact) but I believe there are other tools used (e.g. by insurance companies and others) to recreate what happens after a collision.

JuliaEmelianova, 2021-07-06:
@lemketron Thanks for the explanation

closed 2021-07-06 by EricBoiseLGSVL