lgsvl/simulator #1544: How can I add custom scripts and other plugins?

Cdanzhi, 2021-07-04:
I tried to write a script that will change the weather with the change of player position and load it on an Gameobject. Then I modified the script UIManager that comes with LGSVL to automatically generate this Gameobject after starting to run. There doesn't seem to be any problem with running directly into the test scene. But if I run LGSVL from the Loader scene, Unity will say after loading the scene:"The referenced script on this Behaviour(Gameobject "weathercontroler") is missing!"Then Unity crashed.
How can I avoid this problem? Should I put my script in a particular place?
Also, if I am going to add some other plugins to LGSVL that will run in real time, should the resources of the plugins (such as textures, scripts) be placed in a specific place?
Thanks very much!

EricBoiseLGSVL, 2021-07-06:
If you are adding functionality to simulator you should place those scripts in the source and build new binaries.  We recommend building custom sensors so that you don't need new binaries.  These should be placed in the External/Sensors/YourSensor folder.  All assets pertaining to the sensor logic/art should be in that folder.  Please see docs [here](https://www.svlsimulator.com/docs/plugins/sensor-plugins/).

With the info you have posted, it seems you be doing a few things wrong:
1) Are you adding a gameobject to a scene and not saving, nor rebuilding the map? Since we don't support custom scripts in environments (looking to support sometime soon), the script needs to be in simulator code so Unity can load the script. It also needs new binaries if you want to run the exe.

2) Do you have the script in Assets/Scripts?  Not sure what the test scene is but if Unity can't find the script you reference in UIManager then Unity shouldn't let you compile anything. UIManager is not a good place to add this btw, it's for UI focused logic and you will have issues if you mix logic in places it doesn't belong.  EnvironmentalEffectManager.cs would be best for methods to get data and change weather while the sensor gets position and sets states.

If you could give more precise info on the code you have written I can give better feedback.

Cdanzhi, 2021-07-07:
> If you are adding functionality to simulator you should place those scripts in the source and build new binaries. We recommend building custom sensors so that you don't need new binaries. These [...]

Cdanzhi, 2021-07-07:
> If you are adding functionality to simulator you should place those scripts in the source and build new binaries. We recommend building custom sensors so that you don't need new binaries. These [...]

Thank you very much for your reply and suggestions!
I am currently using a Simulator -- Release202006 and Unity version is 2019.3.15.
I'm trying to load my script using the Sensor Plugin, but I'm running into a new problem.
I downloaded ComfortSensor, but when I imported these files into Unity /...After /External/Sensor/ComfortSensor, there are quite a few errors, as shown in the figure below.

[image]

How should I solve this problem?Thank you very much again!

EricBoiseLGSVL, 2021-07-07:
Be sure that the comfort sensor branch/tag is for your version of simulator.  Don't checkout master for the sensor when your simulator is 06