lgsvl/simulator #1546: EGO incorrectly detects rotation of NPC vehicles at Highway101GLE map

JuliaEmelianova, 2021-07-05:
**Description**
When I run a simulation with random traffic and use Highway101GLE map, EGO incorrectly detects rotation of NPC vehicles:

[image]

I have downloaded HD Map (apollo 50) for Highway101GLE from the Web UI Store and used a script generate_map.sh to generate map files. I use Apollo 6.0 (modular testing) for sensor configuration.

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo version: master-2021-07-01-9120b75d4a
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

JuliaEmelianova, 2021-07-05:
EGO incorrectly detects rotation of random NPC vehicles also when I use Shalun map:

[image]

EricBoiseLGSVL, 2021-07-06:
Hmm, might be a map origin issue.  Thanks for posting.

EricBoiseLGSVL, 2021-09-15:
This has been fixed and will be in our next release.  Thanks for posting

closed 2021-09-15 by EricBoiseLGSVL