lgsvl/simulator #1547: How to Add a New Ego Vehicle

DestinyFrank, 2021-07-05:
I am having problems 

creating a new ego vehicle in the SVL Simulator. I have been following this tutorial ( https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/). After completing the steps for preparing the asset in Maya. The main issue I am having is launching the svl simulator from the unity editor. When that wasn't working I tried to continue with setting up the vehicle in Unity. However, I keep getting error messages when I try to Build the simulator in unity. I have also tried to upload the vehicle as a  unity asset bundle directly into the website, but it does not match the requested type. Any help would be greatly appreciated.
Thank you. 
[image]

heeen, 2021-07-06:
What operating system/platform are you developing on? Is it 32bit or 64 bit?

DestinyFrank, 2021-07-06:
My operating system is 64 bit.

heeen, 2021-07-06:
Windows or Linux?

DestinyFrank, 2021-07-06:
My computer is Windows.

EricBoiseLGSVL, 2021-07-06:
@DestinyFrank hello and awesome you are creating a new vehicle for svl simulator.  It is a complex process and I hope I can help.

A few things before building a vehicle:
- You seem to have a corrupted database.  Go to C:\Users\YOU\AppData\LocalLow\LGElectronics and delete the SVLSimulator folder.  Then download the latest binary and run it.  Link the simulator to the cloud and then run a simple simulation to test issues running the sim (cubetown, jag with only keyboard sensor).  Copy the config.yml file in the binary and add it to the simulator source root folder (unity project) if it is not there.
- You need to run the simulator in the Unity Editor by starting the LoaderScene.scene.  If you complete the above steps, the editor will run the simulator just like a binary.

Let me know how it goes to run the simulator first and then post the errors you are having building. Errors are in the player.log for the binary and in the console window for the Editor.

A few things to check before trying to build again:
1) be sure Linux mono build support is installed in your windows Unity installation.
2) be sure you have the right version installed. 2019.4.18f1 is the version for 2021.1.1 release.
3) be sure you are placing all assets in Assets/External/Vehicles/YourVehicle.
4) vehicle prefab needs to be named YourVehicle.

Try again, post errors and take a screen shot of the vehicle in the Unity Editor with a view of the Hierarchy Panel and a screen shot of the bundle opened with a zip program (the bundle is just a zip and you can open it to see what files are missing).

DestinyFrank, 2021-07-07:
@EricBoiseLGSVL  Thank you for your help. I was able to build the vehicle in Unity and upload it to the simulator website. I am getting an error that the versions are mismatched. I checked the other vehicles in the library and it seems that the rest are for simulator 3 while ours in simulator 5. Is there a way to change the version or do I need to change building it in Unity. 
[image]
[image]

EricBoiseLGSVL, 2021-07-07:
Glad to help.  We have been updating webui all day today. Try to clear cookies, sign out and sign back in.  See if the assets have been updated.

DestinyFrank, 2021-07-12:
Hi, @EricBoiseLGSVL I am trying to get the vehicle to show when I run the simulation. I changed the configuration and I am trying to select a bridge to see if that helps. I tried using the ROS bridge because it is the same as the other vehicle I used. However, when I try to select an autopilot it says that it is not compatible with the selected vehicle's bridge. I tried another bridge-type but it says the same thing. Is there a different bridge I need to use or an update? 
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-12:
Yes we have removed bridges from the source code and posted to github. I recommend creating a new vehicle and use the new bridge in webui.  Sorry for the confusion.

DestinyFrank, 2021-07-12:
Okay, thank you for letting me know. I am still a little confused about where is the WebUI is? Is there a setting when you upload the AssetBundle to the website or is that in Unity? After trying again I think I fixed the issue with the ROS bridge. I was able to select  Autoware.AI v1.12 for the Autopilot and it was able to run the simulation. I am still not able to see our vehicle in the simulation view. 
[image]
[image]

EricBoiseLGSVL, 2021-07-12:
hmm it is loaded.  What is the scale? Can you post the bundle here?

DestinyFrank, 2021-07-13:
My original scale was 1 I tried I changed it to 2 to see if that would help. The vehicle is still not showing when I run the simulation it does show in the Vehicles Library. 
[image]

EricBoiseLGSVL, 2021-07-13:
Can you run the simulation in the Editor so we can look to see what Unity is spawning?  The camera is centered at the spawn point so it seems the issue is the vehicle itself.  Please run loader, then the simulation and then open AgentManager under SimulationManager to select the vehicle.  Then you can press F to snap to the agent.  Let's get a screen shot of the vehicle in the hierarchy panel.

DestinyFrank, 2021-07-13:
I am not able to run the simulation from the Unity editor. When I  have tried to run the simulation in unity I have gotten errors. I am also having a hard time finding the settings you are talking about in Unity I apologize. 

[image]
[image]
[image]

DestinyFrank, 2021-07-13:
Here is the asset bundle of the vehicle I am using. 
[vehicle_Robot_Medium_ALL_2.zip](https://github.com/lgsvl/simulator/files/6811204/vehicle_Robot_Medium_ALL_2.zip)

EricBoiseLGSVL, 2021-07-13:
First off, don't run editor simulation with developer settings.  Run the LoaderScene.  Then look at the game object that spawns for your vehicle in the Hierarchy Panel while a simulation is running.  We need to see what Unity is spawning as an agent.  That way you can see if the pivot is off or the mesh is missing.

The meshes look ok but we need to see how you placed the pivot point of the mesh and holder object.
Does your pivots look like these?
[image]
[image]

Are the pivots at the bottom of the model, not at the bottom of the chassis?

EricBoiseLGSVL, 2021-07-13:
The issue with your build is that webui is not able to get a version from your locally built binary of simulator.  We are working on a fix and should push very soon.

DestinyFrank, 2021-07-14:
I apologize it been a while since I've used Unity what do you mean by  Run the LoaderScene? Here are images where the pivots on the vehicle is this correct? 
[image]
[image]

EricBoiseLGSVL, 2021-07-14:
The pivot in local space is not right I think.  Look again at the images I posted and the underline in red.  Notice the gizmo is at the bottom of the vehicle, not in the middle.  Vehicles are placed at the spawn point at this position in world space.  So in simulator, the vehicle is placed mid way in the road collision mesh, and falls through.  If you look in the editor when the simulator starts, I bet the vehicle is falling below the map. (Check local space is toggled on in the top left, you cut it out of the image you posted and you can see in mine)

Loaderscene is the scene in the editor you start with play mode to test.  It loads the main menu and you can run simulations just like a binary. Look for it in Assets/Scenes.

Doc [here](https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/) video at 0:19 for origin point.

DestinyFrank, 2021-07-15:
I changed it to Pivot and local and it seems to be at the right spot below the vehicle. I was able to do the find Loaderscene but when I pressed play it said that it was disconnected and that it does not match the linked version 2021.2. Since I changed to the gizmo setting should I try to build the vehicle again and uploaded it to the website again? 

[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-15:
Since the cluster is linked to the binary that is 2021.2 and the editor is technically Development you need to unlink the cluster and link with the editor.  Then 2021.2 and the editor will both work.  You can also add a data_path to the binary so it uses a different simid and cache folder.

DestinyFrank, 2021-07-15:
I deleted that original cluster and connected it to the one in the Loaderscene but it's now saying that is not compatible with the simulation cluster version.
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-15:
You need to replace the old maps/bridge with the new compatible ones in the simulation

DestinyFrank, 2021-07-15:
Which bridges and maps are compatible with the simulation? When I am looking at the other maps it's saying that none of them are. Also, we wanted to find a map that was a city area is there another one like the San Francisco one?

EricBoiseLGSVL, 2021-07-15:
what webui url are you using?  Hit ctrl F5 or delete cookies for the url because you should see updated assets

DestinyFrank, 2021-07-15:
I am unsure about WebUI I did not have the option in the simulator tab in unity which is where the tutorial said it would be. 
[image]
 said it would be.

thisisdavidfrank, 2021-07-15:
@DestinyFrank, WebUI is just a term we use for the the user interface on your browser. Please go to the browser where you are seeing this incompatibility errors, and try a hard-refresh (can be done by using Ctrl + F5). This should clear your local browser cache's stale data and reload it with the compatible versions.

EricBoiseLGSVL, 2021-07-15:
Why are you on Unity Editor 2020.3.12f1?  This is not right.  You need to be on 2020.3.3f1 per our documentation.  I'm guessing you are on wise.svlsimulator then but you can check in the bottom left in loader scene after hitting play mode. You may need to reclone everything because you are using a version of unity that is not supported

heeen, 2021-07-15:
It looks like on wise.svlsimulator.com San Francisco is not marked as compatible with Development & dev.

EricBoiseLGSVL, 2021-07-15:
Darn looks like this needs fixed in the back-end.  @eagafonovlge

eagafonovlge, 2021-07-15:
@DestinyFrank 
What is the exact version of the simulator your build is based on?

Looks like it is not 2021.2 and it's not 2021.1 either, but something in between. Could you please pick 2021.2 version as the base version for your changes?

DestinyFrank, 2021-07-16:
I have tried to download different versions of Unity and the simulator files but I am still really confused. @EricBoiseLGSVL would it be possible to set up a meeting or is there a person I can contact? I have been working on this project for a couple of weeks and I am not sure how to move forward.

EricBoiseLGSVL, 2021-07-16:
- Download Unity Editor 2020.3.3f1, remove versions not supported
- Delete all local AppData folders for simulator
- Clone release 2021.2 simulator into a new folder
- Clone release 2021.2 tags for any open source repos you are using to this new project
- Use wise.svlsimulator.com cloud url
- Remove all assets/clusters/simulations in your library in webui.
- Create a simple simulation with the Jaguar/Borregas/KeyboardSensor
- Run the simulator from LoaderScene in Editor.

Once this runs without errors, we can try to find out why you cannot add a new vehicle.  Because you are using the wrong Unity version, there is no point in screen sharing.  Let's get a clean slate first.

DestinyFrank, 2021-07-19:
@EricBoiseLGSVL I followed those steps and was able to load the simulator into Unity. After doing that, I tried to create a simulation, but it would not like me to select a map, but I could choose a vehicle. It's not giving me any error messages telling me why it's not connecting.  
[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-19:
@DestinyFrank It seems you don't have the assets that supports development.  Any idea @eagafonovlge or @thisisdavidfrank ?

lemketron, 2021-07-19:
> After doing that, I tried to create a simulation, but it would not like me to select a map, but I could choose a vehicle. It's not giving me any error messages telling me why it's not connecting.

You'll need to select a map. Are you saying there were no choices available?  If so then you'll need to add a map to your library. To add the Borregas map, go here:
https://wise.svlsimulator.com/maps/profile/aae03d2a-b7ca-4a88-9e41-9035287a12cc

And click "Add to Library" in the upper right corner. Then you should be able to return to your simulation and select the Map.

EricBoiseLGSVL, 2021-07-19:
It looks like those are added but not compatible with the development version. Those may be the old ones though.  @DestinyFrank please try to remove all maps in your library in webui maps and then ctrl f5 and then add borregas from the public ones.

DestinyFrank, 2021-07-20:
@EricBoiseLGSVL The map says that is it not compatible with the simulation cluster version. Also, I was wondering if you had a chance to try on your end to upload the vehicle with the files I sent? 
[image]
[image]

EricBoiseLGSVL, 2021-07-20:
It seems the the map is not the latest.  I'll ping @eagafonovlge and @thisisdavidfrank on this.  You are on wise.svlsimulator.com and release 2021.2 right?

DestinyFrank, 2021-07-20:
Okay thank you and yes I am on both of those. @EricBoiseLGSVL

eagafonovlge, 2021-07-20:
@DestinyFrank 
Please, unlink the simulator and link it again. It would refresh asset bundle format mapping for your Development simulator version, so 2021.2 release asst bundles would be marked as compatible.

DestinyFrank, 2021-07-20:
Thank you! It is now able to run the simulation in the unity editor. @eagafonovlge @EricBoiseLGSVL 

[image]

eagafonovlge, 2021-07-20:
You are welcome!

closed 2021-07-20 by eagafonovlge

DestinyFrank, 2021-07-22:
@EricBoiseLGSVL @eagafonovlge Thank you for your help. I looked back at the tutorial to upload a new ego vehicle and it is now able to be seen in the simulation.  The only issue I having now is that when I press play the car and the screen seems to freeze. Is there a configuration I need to add or something I missed when creating the assets bundle? 
[image]
[image]
 [image]

EricBoiseLGSVL, 2021-07-22:
When you press play in the UI at the bottom left the simulator freezes? Do you have errors in the console panel?

DestinyFrank, 2021-07-22:
Yes, once I click the play button it freezes. I can move the camera around and click the icons until I press the play button and I am not getting any error messages.

EricBoiseLGSVL, 2021-07-22:
Are you running the simulator in editor from the LoaderScene? How are you running the simulator?  It seems you have the play/pause state inverted.

DestinyFrank, 2021-07-23:
@EricBoiseLGSVL I am running the simulation in the LoaderScene. I am clicking run simulation on the website and then pressing play in the game view.

EricBoiseLGSVL, 2021-07-23:
Did you change any code for state?  The issue I see is that your simulation is paused but the button state is inverted. So when you press play the simulation actually pauses.  Do you have any errors in the console panel? If you have an error then the initialization will fail. Please screen shot the console panel or share your editor log? Another thing to check is what testcase template are you running from the webui?  What type of simulation are you running in the editor?

DestinyFrank, 2021-07-23:
@EricBoiseLGSVL I think that the issue is with the pause button. Once I pressed play I went and pressed the pause button for the game editor and it worked. I can use the arrow keys the move it now and all the wheels are moving in the right way.  Thank you so much for your help.

EricBoiseLGSVL, 2021-07-23:
Odd, you may have a editor setting that pauses on errors.  See [here](http://docs.unity3d.com/Manual/Console.html). Glad to help and that you resolved the issue.

DestinyFrank, 2021-07-23:
@EricBoiseLGSVL Thank you! I turned off the error pause and it works when I press the play button.