lgsvl/simulator #1549: Should random pedestrians be available at all maps or not

JuliaEmelianova, 2021-07-05:
Could you clarify please if random pedestrians are available at all maps or not?

When I try to run a simulation with random pedestrians at Highway101GLE and Shalun maps using "Random Traffic" runtime template, there are no pedestrians visible. At the same time, all looks fine when I try to do this with BorregasAve or SanFrancisco maps. 
When I use "PythonAPI" or "API Only" runtime template, the simulation is stuck due to the method `sim.add_random_agents(lgsvl.AgentType.PEDESTRIAN)` in my code.

I have downloaded HD Map (apollo 50) for Highway101GLE from the Web UI Store and used a script generate_map.sh to generate map files. I use Apollo 6.0 (modular testing) for sensor configuration.

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo version: master-2021-07-01-9120b75d4a
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

EricBoiseLGSVL, 2021-07-06:
@JuliaEmelianova Thanks for the specs, that is very helpful.
Not all maps support pedestrians.  Highway101GLE and Shalun do not. We are working on ways to inform users that certain features are available on certain maps.
GoMentum, Borregas, CubeTown and all Straight1(2) with Pedestrian in the title maps support pedestrians.
Thanks for posting that API hangs on this call we should catch this and report it is unsupported.
Sorry, pedestrian annotation data is not exported in the hdmap data.

JuliaEmelianova, 2021-07-07:
@EricBoiseLGSVL Thanks for the reply

EricBoiseLGSVL, 2021-07-07:
@revati-naik please make a bug for the hang in API with the add agent if no pedestrian annotation is present.  Thanks

joeking11829, 2021-07-08:
> Sorry, pedestrian annotation data is not exported in the hdmap data.

According to the document [here](https://www.svlsimulator.com/docs/simulation-content/map-annotation/#create-pedestrian-navmesh).

Could I add the pedestrian annotation data to the Shalun ?
After that, would it enable the feature of random pedestrians ?

EricBoiseLGSVL, 2021-07-08:
NavMesh is not pedestrian annotation.  It is just the mesh that NavMeshAgents can use to path-find in a map.  See [here](https://docs.unity3d.com/Manual/class-NavMeshAgent.html).
Yes, you can add annotation to Shalun and pedestrians will work there.  I will make a note to annotate for our next release as well.

EricBoiseLGSVL, 2021-07-26:
Added checks for pedestrian and traffic annotation for add random agent api commands.  Thanks @JuliaEmelianova for posting.  This will be in the next release.

closed 2021-07-26 by EricBoiseLGSVL