lgsvl/simulator #1550: How get the map of com.svlsimulator.4?

DZiYou, 2021-07-06:
I use the simulator build from source 2021.2,but when I run it ,there is a error : BUndleFormat: CUbe Town is for bundl version com.svlsimulator.2 but currently running simulator supports only com.slvsimulator.4 bundle version.
I can't found com.slvsimulator.4 map in map store,how could I get it?Thank you .
[image]

zelenkovsky, 2021-07-06:
*IMPORTANT* Please read https://www.svlsimulator.com/news/2021-06-08-weekly-pushes

Do not upload asset bundles built using cutting edge master to https://wise.svlsimulator.com production cloud deployment. Use wise.preview.svlsimulator.com instead.

closed 2021-07-06 by zelenkovsky

DZiYou, 2021-07-06:
> _IMPORTANT_ Please read https://www.svlsimulator.com/news/2021-06-08-weekly-pushes
> 
> Do not upload asset bundles built using cutting edge master to https://wise.svlsimulator.com production cloud deployment. Use https://wise.preview.svlsimulator.com instead.

 So ， How to link 2021.2 SImulator to https://wise.preview.svlsimulator.com ?
          and https://wise.preview.svlsimulator.com  map is com.slvsimulator.3,not equla com.slvsimulator.4
thank you

lokesku, 2021-07-06:
2021.2 is still not released. We have just pushed the spurces, but we will be updating the production and preview deployments this week to add support for 2021.2 release. Please wait for few days till we announce 2021.2 release.