lgsvl/simulator #1551: How should I use the VSE with self-build lgsvl simulator and map

joeking11829, 2021-07-06:
Hi LGSVL Teams,

**Q1:** 
I want to use the VSE to create some scenario.
But I need a self-build simulator and map, because I add some new tags for "Segmentation Sensor".

How should I do ?

**Q2:**
I just want to create some scenario and collect some data from Cyber Bridge.
How could I disable the dreamview api when I run the VSE simulation ?

EricBoiseLGSVL, 2021-07-07:
@joeking11829 Please see the docs on building local binaries [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).
1) Just rebase your changes on the release branch.  VSE must be run on **Linux** only
2) You will need to wait until we support custom runners.  This should release soon.

joeking11829, 2021-07-08:
Thanks for the reply.

I will try the release branch.

closed 2021-07-08 by EricBoiseLGSVL