lgsvl/simulator #1552: EGO detects NPC vehicles which are below the highway bridge as being on the bridge in Apollo Dreamview

JuliaEmelianova, 2021-07-06:
**Description**
When EGO drives on a highway bridge and NPCs drive below this bridge, EGO detects them as being on its path  in Apollo Dreamview.

**Steps to reproduce**
1. Download HD Map (apollo 50) for Highway101GLE from the Web UI Store in SVL Simulator 
2. Use a script generate_map.sh to generate map files
3. Run Apollo with the SVL Simulator
4. Run a simulation with EGO driving on a highway bridge and NPCs driving on a highway below this bridge (it may be random traffic). Use Apollo 6.0 (modular testing) for sensor configuration
5. Open Apollo Dreamview in a browser by navigating to: localhost:8888 
6. Check the EGO vehicle navigation when EGO drives on a highway bridge
**Expected result**
EGO detects NPC vehicles being on the highway bridge and does not detect NPCs below the highway bridge.
**Actual result**
1. EGO detects NPC vehicles being on the highway bridge.
2. EGO detects NPC vehicles which are below the highway bridge as being on the bridge.

[image]

**System details**
Intel(R) Core(TM) i9-7900X CPU @ 3.30GHz
Nvidia GeForce GTX 1080 Ti (GPU memory 11GB)
Ubuntu 20.04.2 LTS
Apollo version: master-2021-07-01-9120b75d4a
Docker version: apolloauto/apollo:dev-x86_64-18.04-20210517_1712
SVL Simulator 2021.1.1
Browser: Chrome

lemketron, 2021-07-06:
It sounds like the vehicle on the highway under the overpass should not be reported as a detected obstacle by the 3D ground truth sensor. We will check that out, thanks for reporting it!

EricBoiseLGSVL, 2021-07-07:
This is not a bug.  Ground Truth sensor is a simple trigger volume that reports all colliders in it's volume.  It never raycasts/checks if anything is in the way.  If you want this functionality you can modify the sensor or create a new sensor that checks.  We have discussed internally to support this but no timeline as of yet.

JuliaEmelianova, 2021-07-07:
@EricBoiseLGSVL Thanks for the information