lgsvl/simulator #1554: LGSVL and Open Simulation Interface (OSI)

ffzero1, 2021-07-06:
Hello, 

Does SVL simulator support open simulation interface (OSI) to work with test management tool such as ECU-Test?

Thanks!

lemketron, 2021-07-06:
SVL Simulator does not support [OSI](https://www.asam.net/news-media/detail/news/bmw-transfers-open-simulation-interface-osi-to-asam/). SVL Simulator was designed for real time simulation and thus sensors are very tightly coupled with the core simulator (though built as separate plug-ins) for high performance and frame rate when simulating complex sensor sets.

Because SVL Simulator is open source and extensible through many plug-in APIs as well as the Python API, it's possible that someone could create an interface between OSI and SVL but I'm not aware of anyone who has done so.

ffzero1, 2021-07-06:
Thank you for the answer.

closed 2021-07-06 by ffzero1