lgsvl/simulator #1555: Custom map asset bundle upload error

vin-cloud, 2021-07-07:
Hello, facing issues with uploading generated asset bundle of custom map.
Process followed as below
a. Created Custom map based on Roadrunner tool
b. Unity import and build
c. AssetBundles generated -> can extract the contents
d. Upload on svlsimulator.com portal -> error as seen below
[image]
AssetBundle file link: http://naver.me/5DHtMdRI

EricBoiseLGSVL, 2021-07-07:
What version of simulator are you building assets with?

vin-cloud, 2021-07-07:
> What version of simulator are you building assets with?
Please find the details below:
[image]

EricBoiseLGSVL, 2021-07-07:
Since you are using the latest release (that really shouldn't have gone out yet). You will need to make a new bundle with the latest simulator source code.  Please build a new bundle and use preview webUI see #1550 .  Sorry about the confusion

vin-cloud, 2021-07-07:
> e build a new bundle and use

The _master_ version?

EricBoiseLGSVL, 2021-07-07:
ah ok, master is latest so you will need a new version of Unity.  Please see the #1550 issue. master is preview and uses the latest changes.  Unity version is 2020.3.3f1.  Not sure what version you were using before because latest release was 2019.4.18?
Release tags are our official release with docs (except for 2021.2 bc it was released early on accident) that state the Unity version
Master is a preview for users that want to see in progress work and only work on wise.preview. If you are not familiar with the simulator I recommend you stick with 2021,1 release and use unity 2019.4.18f1 until our backend is up and running for release 2021.2. Sorry about the confusion.

vin-cloud, 2021-07-07:
@EricBoiseLGSVL Tried generating Asset Bundle using the version you mentioned,
LGSVL: 2021.1 and Unity 2019.4.18f1.
**Issues:**
a. **Cannot generate Asset Bundle (Build has Binary Options (Not Bundles))**
[image]

b. **Error while building** 
[image]

EricBoiseLGSVL, 2021-07-07:
You have an error in the map annotation.  You also need to delete the directional light in scene when you build.
```
var firstPt = lane.transform.TransformPoint(lane.mapLocalPositions[0]);
```
You probably have a lane with 0 local positions.

bluecar12, 2021-07-09:
@EricBoiseLGSVL Hey, I'm having the same issue as referenced here, but I'm using the versions you suggested here: 2021,1 release and unity 2019.4.18f1. To eliminate the possibility that it is my project that is the issue, I am using the CubeTown example project. Do you have any suggestions to solve this? Screenshots below
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-12:
Please post the modified cubetown so I can see if the bundle is correct. Do you have errors when you try to build it?

bluecar12, 2021-07-13:
@EricBoiseLGSVL
I solved the issue and was able to get it uploaded. 
>Do you have errors when you try to build it?
For future reference, fixing all build errors is crucial to getting the assetbundle to upload. This seems fairly obvious in retrospect, but I assumed build errors were somewhat ignorable like other large projects.

Thanks for your help

EricBoiseLGSVL, 2021-07-13:
Ah, good that you solved it.  Yes, nothing will work if you have compile errors. That is why I always pull the console panel right next to the project panel on the bottom of the editor to always see errors.  Warnings and logs are fine but red == must fix :)

EricBoiseLGSVL, 2021-07-13:
@vin-cloud were you able to solve your issue as well?

closed 2021-07-16 by EricBoiseLGSVL