lgsvl/simulator #1556: wise connection closed

hean1, 2021-07-07:
svl in docker (use host network) cannot connect to new webui, log showed "WISE connection closed"

heeen, 2021-07-07:
Can you kindly provide the logfile of `./simulator -logFile output.log` in your environment?

hean1, 2021-07-07:
do you mean this logfile?   

> 
root@gpu181-PowerEdge-R740:~/.config/unity3d/LGElectronics/SVLSimulator# cat Player.log 
Mono path[0] = '/opt/simulator/simulator_Data/Managed'
Mono config path = '/opt/simulator/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'VGA-1': 1024x768 (primary device).
Loading player data from /opt/simulator/simulator_Data/data.unity3d
Desktop is 1024 x 768 @ 60 Hz
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.4.18f1 (3310a4d4f880)
[Subsystems] Discovering subsystems at path /opt/simulator/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil)
[Vulkan init] Physical Device 0x15f1cb0 [0]: "TITAN RTX" deviceType=2 vendorID=10de deviceID=1e02
[Vulkan init] Selected physical device 0x15f1cb0
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=128
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_deferred_host_operations, enabled=0
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[... 189 lines cut ...]
  Thread -> id: 7f98a4ff1700 -> priority: 1 
  Thread -> id: 7f98a47f0700 -> priority: 1 
  Thread -> id: 7f98a3fef700 -> priority: 1 
  Thread -> id: 7f98a37ee700 -> priority: 1 
  Thread -> id: 7f98a2fed700 -> priority: 1 
  Thread -> id: 7f98a27ec700 -> priority: 1 
  Thread -> id: 7f98a1feb700 -> priority: 1 
  Thread -> id: 7f98a17ea700 -> priority: 1 
  Thread -> id: 7f98a0fe9700 -> priority: 1 
  Thread -> id: 7f98a07e8700 -> priority: 1 
  Thread -> id: 7f989ffe7700 -> priority: 1 
  Thread -> id: 7f989f7e6700 -> priority: 1 
[CONN] Connected to WISE.
WISE connection closed 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

[CONN] Disconnecting
Giving up reconnecting. 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

[CONN] Disconnecting
Vulkan PSO: data extracted [new size=3715255, old size=0, incomplete=0]
Vulkan PSO: Saved size[3715255] path[/tmp/LGElectronics/SVLSimulator/vulkan_pso_cache.bin] success=1 headerChanged=1 sizeChanged=1

>