lgsvl/simulator #1558: Rev limiter on the Dallara IL-15?

TrentWeiss, 2021-07-07:
I have successfully spawned a [Dallara IL-15 racecar](https://content.lgsvlsimulator.com/vehicles/dallara-il15/) in the simulator on the [Indianapolis Motor Speedway Map](https://content.lgsvlsimulator.com/maps/indianapolismotorspeedway/) and can drive the car via both keyboard control and via ROS2 messages through a ROS2 bridge.

However, I notice that the vehilce speed that comes off of the Odometry topic I have set up tops out at a value of ~42. I am not sure what unit that is. Is it m/s? If so, that makes no sense.  42 m/s is only ~150 kph, WAY slower than what an IndyLights car can do.  I have also noticed that even when going on a straightaway with no steering command applied, the car will suddenly start to slow down when the vehicle speed exceeds 42 (whatever magic unit that is).  If this really is m/s, then that corresponds to ~90mph, which is a sensible speed limit for a road car, but makes no sense for a race car.  

Is there some sort of built-in rev limiter that is forcing the car to slow down because the sim thinks the vehicle is going at an unsafe speed?

lemketron, 2021-07-07:
Hi @TrentWeiss,

I believe the reason for this is buried deep within [VehicleSMI.cs](https://github.com/lgsvl/simulator/blob/fdea5ad112a02e24c06dac8c934328ec139d0763/Assets/Scripts/Dynamics/Examples/VehicleSMI.cs) as documented here:
https://www.svlsimulator.com/docs/simulation-content/ego-vehicle-dynamics/#simple-model-interface

It's not so much a rev limiter (as I've tried changing that to no avail). The vehicle RPM does not directly translate into moving the car at a particular speed but applies a force at the wheels which is subject to other factors like friction, drag, etc.

This SMI code was created for the original Jaguar vehicle model, and likely never intended to power an IL-15 at speeds over 100mph.  I've been in contact with some folks at Unity to see if they can help us figure out how to fix this and allow the car to go faster. Of course, anyone is welcome to check out the [Jaguar vehicle source](https://github.com/lgsvl/Jaguar2015XE) and let us know if you're able to make it go faster. Access to a private IL-15 repo can be arranged as well for someone seriously interested in looking into this.

Thanks!

TrentWeiss, 2021-07-07:
Yeah that tracks. (Pun probably intended)

Access to the IL-15 repo certainly wouldn't hurt (we can discuss that offline).  But if the issue is the core dynamics modelling of the sim not matching the true dynamics of an IL-15 (by a non-trivial margin), then support from the Unity folks is likely required.

What was curious to me is that sudden drop in speed at 42 m/s (it is m/s right?).  A standard drag model wouldn't do that, and the car was on a flat portion of the track going straight, so I don't think friction is the culprit either.

EricBoiseLGSVL, 2021-07-07:
I wonder if this is causing the issue?
```
RB.AddForce(-AirDragCoeff * RB.velocity * RB.velocity.magnitude); //air drag (quadratic)
RB.AddForce(-AirDownForceCoeff * RB.velocity.sqrMagnitude * transform.up); //downforce (quadratic)
RB.AddForceAtPosition(-TireDragCoeff * RB.velocity, transform.position); //tire drag (Linear)
```
It would make sense as the velocity increases, the AddForce cancels the force added from the wheels rotation

TrentWeiss, 2021-07-07:
> It would make sense as the velocity increases, the AddForce cancels the force added from the wheels rotation

Yeah that makes sense, but wouldn't that result in a steady increase in speed that tapers off as aero drag + tire drag starts to equalize the force from the wheel rotation?

I am seeing a sawtooth pattern where the speed steadily increases up to 42 m/s and then suddenly drops for no immediately obvious reason followed by a steady increase up to 42 m/s again (rinse and repeat).

I'll generate some plots next time I have access to my simulator PC.

TrentWeiss, 2021-07-07:
Regardless, an IL-15 engine has WAY more than enough power to overcome drag at only 150 KPH.  Perhaps the drag model parameters are incorrect?

EricBoiseLGSVL, 2021-07-07:
hmm, yes I think you are right.  The other suspect logic is this:
```
if (CurrentRPM / MaxRPM < ShiftDownCurve.Evaluate(AccellInput) && Mathf.RoundToInt(CurrentGear) > 1)
{
    GearboxShiftDown();
}
```
I'd put a debug log to see if the sawtooth pattern lines up with this shift down.

TrentWeiss, 2021-07-09:
> I'd put a debug log to see if the sawtooth pattern lines up with this shift down.

I will do that.

A couple other things that might not be correct:
1. The torque-RPM curve on an IL-15 engine
2. The gear ratio(s) on the simulated gearbox
3. The drag coefficients

Are these things defined in a separate repo (perhaps the private one that @lemketron mentioned)?

EricBoiseLGSVL, 2021-07-12:
Yes, the repo has the public vars that set this except drag. That's in the VehicleSMI,cs

mslavescu, 2021-07-27:
@TrentWeiss did you have any luck in improving the IL-15 speed?

TrentWeiss, 2021-07-30:
> @TrentWeiss did you have any luck in improving the Dallara IL-15 max speed in SVL?

Not yet. Still need support from the Unity folks.

@lemketron, any update on whether Unity will have someone help us on this?

KyleRAnderson, 2021-09-18:
Has there been any progress on this one? We're also a little limited by this issue, is there at least a temporary workaround perhaps?

EricBoiseLGSVL, 2021-09-20:
We have been able to fix this issue by using Vehicle Physics Pro.  It seems that this is an issue with the dynamics model.  The jaguar dynamics is not suitable for the Dallara.  We are currently seeing how we can release as open source without violating VPP licensing.

KyleRAnderson, 2021-09-21:
Thanks for the update, glad that you've been able to find a nice fix 😀