lgsvl/simulator #1559: ROS2 bridge not compatible with ROS2??????

TrentWeiss, 2021-07-07:
All of a sudden. I am unable to create a ROS2 bridge for the Dallara IL-15 vehicle. The simulator is telling me that all of the autopilots available on the dropdown are "incompatible with the vehicle's selected bridge".  Including options such as "Other ROS2 autopilot".  I have verified that ROS2 was selected for my sensor configuration's bridge.

ROS2 autopilots are incompatible with a ROS2 bridge? Huh?

This issue has appeared out of nowhere in the past 30 minutes. I have been able to create a ROS2 bridge to/from the sim before on the exact same simulator version.

Screenshots included for clarity.

[image]

[image]

lokesku, 2021-07-07:
We are updating web interface for our new release. Please try after sometime.

lemketron, 2021-07-07:
It's possible you caught us in the middle of a site update where assets were temporarily unavailable.  Can you refresh the page (or maybe even log out and back in again) and see if it still happens?  Thanks, and sorry for the inconvenience.

lemketron, 2021-07-08:
You could also try unlinking your simulator (menu in the upper right corner) and then (make sure simulator is online and then) click "link to cloud" again, and make a new cluster for it and see if that works? You'll also need to update your simulations to use the new cluster.

MaxandreOgeret, 2021-07-12:
Same problem here but with ROS1. I cannot use the simulator anymore...

[image]

[image]

Here autoware is selected as the bridge and I can't run autoware. Seems like a bug.

TrentWeiss, 2021-07-14:
The issue persists for me as well.  This is definitely not "answered".

Maestroxr, 2021-07-17:
I am facing this issue as well, cannot create any combination of vehicle/map that uses the ROS2 bridge.