lgsvl/simulator #1561: I got an error when i use "Lane Line Detector" after i remove a BoundryLine from annotation.

joeking11829, 2021-07-08:
Hi LGSVL Teams,

I remove the "BoundaryLine_221" from the Shalun map, because it is an incorrect annotation.

[image]

After that, I use the LaneLine Detector to re-generate lane-line information.
But I got an error
[image]

I add some debug message, and I found that it seems to leave an null object.
[image]

How could I completely remove the BoundaryLine and let "LaneLine Detector" works correctly.

Thanks

luqiang21, 2021-07-08:
Looks like "BoundaryLine_221" is the right boundary line of a lane. You need to find out that lane and make sure to have the correct right boundary line for that lane after you remove "BoundaryLine_221".

joeking11829, 2021-07-09:
> Looks like "BoundaryLine_221" is the right boundary line of a lane. You need to find out that lane and make sure to have the correct right boundary line for that lane after you remove "BoundaryLine_221".

Thanks for the reply.

What is the meaning of left/right boundary line ?

luqiang21, 2021-07-09:
> BTW, What is the meaning of left/right boundary line ?

The region between left and right boundary line is the actual area of that lane. The boundary lines may be shared with two adjacent lanes.

joeking11829, 2021-07-12:
> > BTW, What is the meaning of left/right boundary line ?
> 
> The region between left and right boundary line is the actual area of that lane. The boundary lines may be shared with two adjacent lanes.

I got it.
Thank you.

closed 2021-07-12 by joeking11829