lgsvl/simulator #1562: Self-build simulator cannot connect to WebUI server

joeking11829, 2021-07-08:
I found that self-build simulator cannot connect to WebUI server.
But I could connect it yesterday.

[image]

lemketron, 2021-07-08:
Could you try unlinking your simulator (menu in the upper right corner) and then (make sure simulator is online and then) click "link to cloud" again, and make a new cluster for it and see if that works? You'll also need to update your simulations to use the new cluster.

zelenkovsky, 2021-07-08:
If you are modifying the simulator code, making it incompatible with the official release or/and build your own assets, please update ```config.yml``` to use https://wise.preview.svlsimulator.com

Read more here: https://www.svlsimulator.com/news/2021-06-08-weekly-pushes

joeking11829, 2021-07-09:
> Could you try unlinking your simulator (menu in the upper right corner) and then (make sure simulator is online and then) click "link to cloud" again, and make a new cluster for it and see if that works? You'll also need to update your simulations to use the new cluster.

I unlink the cloud and create a new cluster.
I found that when I use self-build simulator to create a new cluster, the simulator version is empty.
[image]

And when I create a new simulation with the cluster, It show some errors.
[image]
just like the issue #1566 

But when I use the official release simulator (svlsimulator-windows64-2021.2), It will show the simulator version.
[image]
Everything will be fine with the Cluster which has simulator version.

Any idea to fix that ?
Thanks.

EricBoiseLGSVL, 2021-07-09:
@thisisdavidfrank

joeking11829, 2021-07-14:
reference to #1574 .
Now I can create new cluster for self-build simulator ( Simulator version: dev ).
[image]

Thanks for fixed the issue.

closed 2021-07-14 by joeking11829