lgsvl/simulator #1564: Disconnected: Simulator version '2021.2' dose not match to the linked version 'Development'(400)

blxl909, 2021-07-08:
When I use the linux release version of simulator, (I have already signed in) after I click the button "Go Online", I get the note "Disconnected: Simulator version '2021.2' dose not match to the linked version 'Development'(400)", how can I deal with it? (I can Go Online when I use the Windows version )

lemketron, 2021-07-08:
Can you (make sure simulator is online and then) try unlinking your simulator (menu in the upper right corner) and then click "link to cloud" again, and make a new cluster for it and see if that works? You'll also need to update your simulations to use the new cluster.

blxl909, 2021-07-09:
Infact when I open the simulator,the status is "Offline". And I can't make a new cluster,  cause the simulator will aks for downloading a release version, but I already download one witch match the platform yesterday, so I am sure the simulator is on the latest version. Maybe there is some ploblem about linking the simulator with  cluster?

lemketron, 2021-07-09:
You should either remove the Simulator folder (at `~/.config/unity3d/LGElectronics/SVLSimulator/`) or click "unlink" in the simulator main window drop down menu in the upper right corner: 
[image]

Once you have unlinked the simulator, then select "go online" and then click the "link to cloud" button in the main screen. This will take you to the browser where you can create a new cluster with a new name. The old cluster can later be removed from the Clusters view, but I recommend NOT doing that until you've either updated or deleted old simulations pointing to the old cluster.

blxl909, 2021-07-09:
Thanks, that works. But I still get a problem when I open the VSE that the map can't be loaded, the note is "BundleFormat: SingleLaneRoad (9c2be962-25d9-452d-98ce-30d4818dc879) is for bundle version com.svlsimulator.2, but currently running simulator supports only com.svlsimulator.4 bundle version", how can I fix that? (I don't know how to get the com.svlsimulator.4 bundle version)

lemketron, 2021-07-09:
Try removing the SingleLandRoad map from your library, then go to the one in the store and add it again. It has a different ID and should be compatible with com.svlsimulator.4 bundle version:
https://wise.svlsimulator.com/maps/profile/a6e2d149-6a18-4b83-9029-4411d7b2e69a/
https://wise.svlsimulator.com/maps/profile/a6e2d149-6a18-4b83-9029-4411d7b2e69a/edit/assets