lgsvl/simulator #1566: Sensor is not compatible with the simulation cluster's version

ByLiZhao, 2021-07-08:
I try to create a simple simulation with  Keyboard Control Sensor, but keep getting the following:

>      Sensor is not compatible with the simulation cluster's version

And my simulation is marked "invalid" with red fonts.

lemketron, 2021-07-08:
Can you try unlinking the simulator (menu in upper right corner of Simulator window) and re-linking to a new cluster?  Then update your simulation to use that new cluster and see if that corrects the problem?  We just released a new version 2021.2 of SVL Simulator yesterday and the corresponding web site update might have caused this issue for you. Sorry for the inconvenience!

ByLiZhao, 2021-07-08:
I am Ubuntu 20.04. I have tried everything that I can think of:

-  Using the pre-compiled binary of the simulator, compiling from source.
- Using release 2021.1 and   using 2021.2.
- Deleting everything and reinstall all related stuff.
- Unlinking and relinking.
- Deleting old cluster and creating new ones.
- registering new user account.
- Deleting vehicles, maps, sensors and adding them back from store.
- Rebooting my computer,
- etc..
Nothing has helped.
Since the new release is released, not only the new release does not work, release2021.1 stops working too.

revati-naik, 2021-07-08:
@ByLiZhao Did you try deleting the SVLSimulator folder which has the data files and the log files? On Ubuntu you would find it at ~/.config/unity3d/LGElectronics/SVLSimulator

DZiYou, 2021-07-09:
I get the same problem and tried to delete the folder ~/.config/unity3d/LGElectronics/SVLSimulator.
 But there is no solution. 
What should I check?

ByLiZhao, 2021-07-09:
> @ByLiZhao Did you try deleting the SVLSimulator folder which has the data files and the log files? On Ubuntu you would find it at ~/.config/unity3d/LGElectronics/SVLSimulator

Yes, It does not help too.

ByLiZhao, 2021-07-09:
Is there anyway to run a simulation from command line without the WEB UI? So that I can debug the problem.
One thing I don't understand. In my case, the "cluster" is just the same PC where the simulator is built and launched. 
How does the "cluster" has a version number?

lemketron, 2021-07-09:
The first simulator of the cluster defines the version of the cluster; we do not allow (a) mix versions in a single cluster and (b) run a different version of the simulator under the same simulator uuid (which might happen if you upgrade the simulator and run it without deleting the old simulator data folder).

ByLiZhao, 2021-07-09:
Your website is now a single point of failure. If I am not mistaken, when your website goes down, or something goes wrong with your server-side setting, no user can login "wise.svlsimulator.com" to run their simulations. 

Plz consider provide a standalone simulator that can run simulations without an internet connection, using the local machine where the binary is launched to run them. If there are some configuration files, put them in a local folder, so that users can view and modify them if necessary. Right now, too much things are hidden behind the WebUI.

vijeshkumarraju, 2021-07-09:
I had the same issue. I have tried unlinking the simulator. Re-Created the cluster with a different name and re-linked it to the newly named cluster. It worked for me then

DZiYou, 2021-07-09:
> I had the same issue. I have tried unlinking the simulator. Re-Created the cluster with a different name and re-linked it to the newly named cluster. It worked for me then

I tried this method, but it didn't work. I created a new blank sensor configuration, which does not add any sensors, can run. But adding configuration is not OK.
[image]

joeking11829, 2021-07-09:
> The first simulator of the cluster defines the version of the cluster; we do not allow (a) mix versions in a single cluster and (b) run a different version of the simulator under the same simulator [...]

For Linux: ~/.config/unity3d/LGElectronics/SVLSimulator

Where is the "old simulator data folder" in Windows System ?

Thanks.

lemketron, 2021-07-09:
For Windows, remove the `SVLSimulator` folder from: `%APPDATA%\..\LocalLow\LGElectronics\`

risk-v, 2021-07-09:
Hi everyone,

Same problem here. I am running Ubuntu 18.04 and running the Simulator from Unity. I have to do run it from the source because I have some custom sensors. It always worked until the new version. Here you have a picture. In my case is complaining about ROS:

[image]

I Tried deleting:  ~/.config/unity3d/LGElectronics/SVLSimulator
- Unlinking
- Creating New cluster and Simulator...
but still getting the error. 
I cloned the 2021.2 branch from github. I have the suspicion it doesn't work because the clusters created from the Simulator have "Development" as version and ROS plugin requires 2021.1, 2021.2. Apparently, is not compatible with Development anymore? Any ideas? Can we change that manually? Thanks in advance.

risk-v, 2021-07-10:
> Hi everyone,
> 
> Same problem here. I am running Ubuntu 18.04 and running the Simulator from Unity. I have to do run it from the source because I have some custom sensors. It always worked until the new version. Here you have a picture. In my case is complaining about ROS:
> 
> [image]
> 
> I Tried deleting: ~/.config/unity3d/LGElectronics/SVLSimulator
> 
> * Unlinking
> * Creating New cluster and Simulations...
>   but still getting the error.
>   I cloned the 2021.2 branch from github. I have the suspicion it doesn't work because the clusters created from the Simulator have "Development" as version and ROS plugin requires 2021.1, 2021.2. Apparently, is not compatible with Development anymore? Any ideas? Can we change that manually? Thanks in advance.

Actually, changing one of the options in config.yml to "wise.preview.svlsimulator.com" fixed my issue.

sparklin1999, 2021-07-10:
I had same problem, and I can run the simulation only when using a blank sensor.

risk-v, 2021-07-10:
@sparklin1999 Did you fix it by editing the config.yml file?

DZiYou, 2021-07-11:
> Actually, changing one of the options in config.yml to "wise.preview.svlsimulator.com" fixed my issue.

Did you means change " simulator/config.yml.template" file , change the "cloud_url:" to "wise.preview.svlsimulator.com" ?

sparklin1999, 2021-07-11:
> @sparklin1999 Did you fix it by editing the config.yml file?

Hi, it seems that if I build svl from source, there only is config.yml.template. If I download the simulator directly, I get "config.yml" file.
And, the cloud_url in config.yml.template is already "https://wise.preview.svlsimlator.com".
[image]
 I also tried to change it to "wise.svlsimulator.com", but that didn't work.

sparklin1999, 2021-07-11:
> @sparklin1999 Did you fix it by editing the config.yml file?

Hi, it seems that if I build svl from source, there only is config.yml.template. If I download the simulator directly, I get "config.yml" file.
And, the cloud_url in config.yml.template is already "https://wise.preview.svlsimlator.com".

 I also tried to change it to "wise.svlsimulator.com" just now, but that didn't work either.

mslavescu, 2021-07-12:
@sparklin1999 if you build from source master branch, you need to use https://wise.preview.svlsimlator.com, just rename the template to config.yml and place it in same folder where the built simulator executable is.
You can also provide the URL on the cmd line with --cloudurl:
https://www.svlsimulator.com/docs/user-interface/config-and-cmd-line-params/#configuration-file

mslavescu, 2021-07-12:
@lemketron  I had to create a new account to be able to run again simulations, had the same issue as other people in this thread.

On my original account I get this error on a new sensor config:
[image]

I removed all clusters and simulations but didn't make any difference.

How can I recover that account, to be able to run simulations again on it?

EricBoiseLGSVL, 2021-07-12:
The issue is being fixed now, it only effects users that are building binaries locally. Sorry for the confusion. We should have a fix very soon.

mslavescu, 2021-07-13:
Thanks @EricBoiseLGSVL, looks good now, I can create new configs on original account.

EricBoiseLGSVL, 2021-07-14:
No problem, glad to help. Sorry about the issues.

dheerajakula, 2021-07-14:
Hi is the issue resolved I am getting the same error for CyberRT

[image]

I am using a development cluster. should I change the cloud url to "wise.svlsimulator.com".

EricBoiseLGSVL, 2021-07-14:
yes, use wise.svlsimulator.com.  Be sure to delete appdata folder before you run the cloud url

dheerajakula, 2021-07-14:
Hi @EricBoiseLGSVL I still have the same issues.

I can see that development version has been added to Keyboard control sensor but not for Cyber RT. I think this may be the issue for the sensor to be not compatible with simulation clusters version. Is there a way in which I can mask my development version as 2021.1.

[image]

[image]

EricBoiseLGSVL, 2021-07-15:
@thisisdavidfrank @eagafonovlge

eagafonovlge, 2021-07-15:
@dheerajakula

Which exactly simulator version do you build?

dheerajakula, 2021-07-15:
@eagafonovlge yes its version 2021.1 I did not build it, I am running it from the unity editor.

dheerajakula, 2021-08-18:
we can use cyberRT bridge in development versions by by building a bridge plugin [https://github.com/lgsvl/simulator/issues/1684#issuecomment-900714008](https://github.com/lgsvl/simulator/issues/1684#issuecomment-900714008)

EricBoiseLGSVL, 2021-08-18:
@dheerajakula we have an issue with bridges with 2021.2.1 and are working on a new release.  Please use development or 2021.2.2 when it releases in the next few days.