lgsvl/simulator #1567: Developer mode Cluster Master tip: Firewall

RobShewagaUHN, 2021-07-08:
Just a tip for everyone: I was having trouble running an offline cluster where the Unity Editor was the master and a remote machine was the client. The remote machine was never able to connect to the Editor, but if I ran the standalone build as the master, it connects fine.

Make sure you don't have any blocking _Inbound Rules_ in _Windows Firewall with Advanced Security_ for the Editor version you're running. Clearing a blocking rule there got it working for me.

EricBoiseLGSVL, 2021-07-09:
Thanks for this post.  I'm sure this will help many users.  You rock.

lusizin, 2022-03-15:
No Comment.

lusizin, 2022-03-15:
No comment

lusizin, 2022-03-15:
No Comment

lusizin, 2022-03-15:
No Comment

lusizin, 2022-03-15:
No Comment

lusizin, 2022-03-15:
No Comment at the moment

lusizin, 2022-03-15:
No Comment to add

lusizin, 2022-03-15:
No Comment

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
2022

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
### ###

lusizin, 2022-03-15:
### ###

lusizin, 2022-03-15:
### ### ### ###

lusizin, 2022-03-15:
### ### ### ###

lusizin, 2022-03-15:
### ### ### ### ### ### ###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
[](url)

lusizin, 2022-03-15:
- [ ]

lusizin, 2022-03-15:
[](url)

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

lusizin, 2022-03-15:
###

[41 more replies]

closed 2022-03-16 by EricBoiseLGSVL