lgsvl/simulator #1568: Keyboard Control Sensor  - Sensor is not compatible with the vehicle's version

nparkstar, 2021-07-09:
All simulations are invalid with message "Keyboard Control Sensor -Sensor is not compatible with the vehicle's version" after using new version(2021.2).
Anybody tell me the way to fix it please?

JWishere, 2021-07-09:
Did you set bridge to any thing but no-bridge? In my case, I could solve that by setting ROS2 brige and ROS2 autopilot. I don't know why.

zelenkovsky, 2021-07-09:
Please provide Keyboard Control Sensor version, version of your Vehicle asset bundle and Version of the Simulator you are using. Make sure your final sensor configuration is compatible with at least one version of the simulator.

What assets are default and what assets were built custom?
Was simulator re-built or you are using pre-built binary?

nparkstar, 2021-07-09:
> Please provide Keyboard Control Sensor version, version of your Vehicle asset bundle and Version of the Simulator you are using. Make sure your final sensor configuration is compatible with at least [...]

Thank you for your answer.
I used binary simulator(svlsimulator-windows64-2021.2.zip).
I just used pre-configured simulation(Local-Random: Cubetown (Manual Drive)) with changing cluster(my cluster) and Sensor Configuration to Keyboard Control.

That's all.

EricBoiseLGSVL, 2021-07-09:
Be sure to clear local assets with clear cache to make sure you aren't loading an old asset.

nparkstar, 2021-07-10:
> Be sure to clear local assets with clear cache to make sure you aren't loading an old asset. It's in the main menu in the top right drop down.

-> I tried, but it didn't work. Did you run simulator with 'Keyboard Control' ?
I want to know everybody run with 'Keyboard Control' successfully on release-2021.2.

nparkstar, 2021-07-10:
[image]

I see this message always.

Maestroxr, 2021-07-11:
I am having the same issue as described by 'nparkstar'.

EricBoiseLGSVL, 2021-07-12:
You need to remove the old sensors from the vehicle configuration.  Sorry, we had some issues migrating with the latest release. Just create a new vehicle and a new sensor config with the updated sensors.  If you clear cache, you need to update webui to use new sensors.  We are working to fix this issue for next release.

nparkstar, 2021-07-13:
> You need to remove the old sensors from the vehicle configuration. Sorry, we had some issues migrating with the latest release. Just create a new vehicle and a new sensor config with the updated [...]

Thank you for your answer, and I hope you fix it asap.
But I don't understand exactly what you said.
Would you tell me more details?

EricBoiseLGSVL, 2021-07-13:
Are you building a new binary?  If so you will need to wait until we can release a fix for users that are building binaries locally.
If not, you need to unlink the simulator in webui and then run the simulator and relink.  This should fix the issue.  Be sure your old vehicle config is not used, this is what I am going through.  Make a new vehicle and add the latest sensors from webui.

DZiYou, 2021-07-13:
> Are you building a new binary? If so you will need to wait until we can release a fix for users that are building binaries locally.
> If not, you need to unlink the simulator in webui and then run the [...]

Sorry.
I made the following attempts:
Step 1: delete all simulation, clusters, vehicle and plugins from webui.
Step 2: close webui and SVL simulator.
Step 3: delete the folder "~ /. Config / unity3d / lgelectronics / svlsimulator"
Step 4: start the downloaded svlsimulator 2021.2 (the downloaded binary file, not the source code)
Step 5: rerun link to cloud, add new vehicles that have never been used before, add new sensor configuration, including keyboard control.
But there is no solution. My question is, why can't I download the binary source code? Do I need to wait for the server to update?

EricBoiseLGSVL, 2021-07-13:
Wait, you can't run the binary source code?  Please post your player.log here so I can see the errors

DZiYou, 2021-07-14:
> Wait, you can't run the binary source code? Please post your player.log here so I can see the errors

There seems to be no problem now. But I didn't do anything, I just repeated the previous operation.
thank you!

EricBoiseLGSVL, 2021-07-14:
back-end pushed a fix :)

nparkstar, 2021-07-14:
> back-end pushed a fix :)

Thank you very much! 
It works!

closed 2021-07-14 by nparkstar