lgsvl/simulator #1571: Failed to launch TestCase runtime

blxl909, 2021-07-12:
When I use the runtime template Python API, I get the error Failed to launch TestCase runtime 
[image]
[image]

maybe I should modified the code?

closed 2021-07-12 by blxl909

Gastastrophe, 2021-07-17:
Did you close this because you fixed the issue? I am having the same problem

blxl909, 2021-07-19:
> Did you close this because you fixed the issue? I am having the same problem
 Yes, I fixed it. But this error may result from many reasons. At this time , it's because I use the offline cluster. But if you are using windows platfrom, since the simulator folder doesn't have a folder named TestCase, it will also cause this error. So you should using Linux instead. Maybe there still other reason, but that what I know for now.