lgsvl/simulator #1572: how to record video?

rajeshjoshi8792, 2021-07-12:
i want to record video for scenario that i created using api. so how i do that? i added video recording sensor in library. but i dont know how to use it. please guide me.

revati-naik, 2021-07-12:
Hi @rajeshjoshi8792, 
To be able to use the video recording sensor, you will need to create a new sensor configuration/clone an existing sensor configuration with the sensors you need for your simulation. To this sensor configuration, you need to add the video recording sensor. Then you can capture the sensor configuration ID for this one and replace the existing vehicle asset ID with this new sensor configuration ID in the API script. 

Hope this helps. Please let us know if you have any further questions.

rajeshjoshi8792, 2021-07-12:
Im not able to find video recording sensor repository in github. Can you send me link to clone repository?

EricBoiseLGSVL, 2021-07-12:
It is not public.  Just use the sensor and you will get a recording after a simulation in AppData folder you set

rajeshjoshi8792, 2021-07-13:
> It is not public. Just use the sensor and you will get a recording after a simulation in AppData folder you set

Which code i write for starting recording? And which sensor i used?

EricBoiseLGSVL, 2021-07-13:
VideoRecordingSensor is the sensor that will record video.  You can also use a camera sensor and collect images from the sensor with the bridge and then just compress once done.

zelenkovsky, 2021-07-14:
Add following [Video Recording Sensor](https://wise.svlsimulator.com/plugins/profile/06aef741-23e6-444b-b6d9-18f84f9ace06) to your sensor configuration and run the simulation.

Once the simulation is over, you will find <Simulator Data Folder>/Videos/xxxxxxxxxxxxxxxxxx.mp4 file on your computer.

enginlien, 2021-08-05:
> Add following [Video Recording Sensor](https://wise.svlsimulator.com/plugins/profile/06aef741-23e6-444b-b6d9-18f84f9ace06) to your sensor configuration and run the simulation.
> 
> Once the simulation is over, you will find /Videos/xxxxxxxxxxxxxxxxxx.mp4 file on your computer.

@zelenkovsky 
Could you please describe more detail? I added a Video Recording Sensor but can't find the mp4 file.

EricBoiseLGSVL, 2021-08-06:
It is in your app data folder.  See [here](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html). It can also be set from config.yml and then it would be at the path you specified.

enginlien, 2021-08-06:
> 它位於您的應用程序數據文件夾中。見[這裡](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html)。它也可以從 config.yml 設置，然後它會在你指定的路徑上。

@EricBoiseLGSVL 

Sorry i still didn't understand,
Could you give me more detail?
I played the file, but I didn’t see the video or images.

[image]

[image]

EricBoiseLGSVL, 2021-08-06:
No video recording sensor is the sensor we created that places video in the data folder.
[image]

Cyber recording is not what we are describing. From your screen shot, you are trying to view a video from recorded images with bridge, this is not video.  You would need to take your images that you save from the topics and use something like ffmpeg to compress into a video file.