lgsvl/simulator #1573: Problem in using  runtime template Python API

blxl909, 2021-07-12:
When I use the runtime template Python API, I got the error

Run python script scenario.py Traceback(most recent call last): File "scenario.py" ,line 8, in <module> sim.load(os.environ.get(LGSVL__MAP")) File "/usr/local/lib/python3.6/sitepack......
line 23, in new_f return f(*args,**kwargs)File "/usr/local/lib/python3.6/sitepack......

the code I use is from https://www.svlsimulator.com/docs/running-simulations/python-runtime/#reading-logs
Example Python Script  
I have BorregasAve map and the Lincoln2017MKZ vehicle to be added to my library, how can I fix that problem?

revati-naik, 2021-07-12:
Hi @blxl909, 
Please refer to the updated Python API script [(10-npc-follow-the-lane.py)](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/10-npc-follow-the-lane.py) to test this scenario.

blxl909, 2021-07-13:
I know that because my python version is too old ,so update python will fix that problem

closed 2021-07-13 by blxl909