lgsvl/simulator #1574: Can't build 2021.2 version

duongtran11, 2021-07-12:
I just cloned release-2021.2 from github
Trying to build it from source. Link it to cloud. Create a new cluster
[image]
But Simulator Version is empty
Then I got these errors
[image]
Seems like the simulator version I built was not 2021.2 although I cloned release-2021.2

EricBoiseLGSVL, 2021-07-12:
We have an issue with users running binaries they built.  We are making fixes right now and should release soon.

duongtran11, 2021-07-13:
> We have an issue with users running binaries they built. We are making fixes right now and should release soon.

Thank you!

zelenkovsky, 2021-07-14:
This issue is supposed to be fixed by now. Could you please confirm and close this issue?

duongtran11, 2021-07-14:
I got it running but one more question:
Why does WebUI still not recognize the new cluster I created ("2021.2?" in the picture) as 2021.2 version? Since I built 2021.2 simulator then link it to cloud then created that cluster "2021.2?", the cluster version supposes to be 2021.2, isn't it?
[image]
Beside that, I couldn't add this simulator to 2021.2 cluster I created before ("cluster_moi" in the picture) with the standalone ready-to-run simulator I cloned from github (not the one I build)
[image]

EricBoiseLGSVL, 2021-07-14:
Because the version you have made is no longer 2021.2.  You have made a custom binary and since users can change anything they want, it is not 2021.2.  This last release was rough because we are trying to make it easier for users to have compatibility but also restrict in webui instead of errors in simulator.
You can't add to cluster_moi because it had the same sim_id from the data_path.  If you change the path for binaries and editor, you can have more clusters active.

closed 2021-07-14 by EricBoiseLGSVL