lgsvl/simulator #1576: Application of ros2 bridge in radar sensor

DZiYou, 2021-07-13:
I use 2021.1 source code, ready to add ros2 radar sensor.
I found 
"public static lgsvl. Detected radar object array convert from (detected radar object data)" 
in the file ros2conversion.cs.
What we need to do is to add 
"regpublisher < detected radarobjectdata, lgsvl. Detected radarobjectdata array > (plugin, ros2conversion. Convertfrom)" 
in ros2bridgefactory.cs Is that right?
1.what else do I need?
2.In the source code of 2012.2, assets / scripts / bridge / ros2 is not found. Do you need to add the ros2 folder?

EricBoiseLGSVL, 2021-07-13:
In release 2021.2, we have moved bridges out of core code so users can create custom bridges.  You need to clone the ROS2 bridge and add the changes, build a new bridge plugin, remove the default one we provide and upload yours to webui.  Then set simulations and vehicles to use your bridge, not ours.
The limited code you posted looks right.  @hadiTab ?

DZiYou, 2021-07-14:
Thank you for your reply, but I still have a few questions.

1. In version 2021.1, since the ros2 bridge is added in the source code by default, why add it to the library from the store. And I found that the version of ros2 in the store does not contain "dev". Maybe for this reason, I modified the ros2 part of 2021.1 source code, but I still can't use it.
[image]

2. I cloned the source code of ros2 version 2021.2, put it into the 2021.2 project, and modified the ros2bridgefactory.cs file as above. But there are two errors as follows:
“Assets/External/Bridges/ROS2/Ros2BridgeFactory.cs(36,91): error CS0407: 'DetectedRadarObjectArray Ros2Conversions.ConvertFrom(DetectedRadarObjectData)' has the wrong return type”
“Assets/External/Bridges/ROS2/Ros2BridgeFactory.cs(36,51): error CS0246: The type or namespace name 'Lgsvl' could not be found (are you missing a using directive or an assembly reference?)”
[image]

Do I need to clone other files to put in? For example, the data folder?

EricBoiseLGSVL, 2021-07-14:
1) this was a fake asset, it just pointed to the local bridge in 2021.1, now it is an actual asset.  This is to support users that have custom bridges. Dev was an error from back end not recognizing local binaries and we will have a fix soon.

2) not sure bc I can't see wait you wrote but looks like you need the correct return type and need a using or update the asmdef.  You shouldn't need anything cloned besides the repo.  This is an issue with the code you added.  Seems just a missing conversion or type.  Just follow the errors and look at the other conversions for help.

DZiYou, 2021-07-14:
Thank you. Now it's ready to run。If others have the same needs, they can refer to it #1085

closed 2021-07-14 by DZiYou