lgsvl/simulator #1579: Cannot add cluster

krtt, 2021-07-13:
After trying to add one more PC to my clusters that are able to run the SVL simulator, I got an error saying that the limit of clusters is reached and cannot add a new one. After deleting several of my old clusters and trying again, the problem persists exactly the same. Tried logging out and back in, no change, still the same message.

EricBoiseLGSVL, 2021-07-13:
This is a bug when you reach max it won't decrease when you remove.  We are looking into this and thanks for posting. @thisisdavidfrank

thisisdavidfrank, 2021-07-14:
@krtt, we are unable to reproduce this issue on the latest release. Can you kindly refresh your browser and try again, please?

If you are still unable to create a new cluster after deleting the older ones, then please send us your email address so that we can take a look at your account more closely to debug it.

Note: If you are not comfortable on writing your email address on GitHub, please send us the email through the 'Contact Us' form on our main website: https://www.svlsimulator.com/.

krtt, 2021-07-19:
Yes, I have refreshed my browser, cleared cache, logged in and out, tried different browsers, tried different computers. Still cannot add cluster even after deleted some older ones. I sent you my email address via the Contact Us form on the svlsimulator.com website. Thanks a lot for looking into this.

EricBoiseLGSVL, 2021-07-19:
@krtt Thank you and we apologize for the issues, we'll get this fixed.

eagafonovlge, 2021-07-20:
@krtt 

Could you please send me your SVL Simulator account email address to eugene.agafonov (a) lge.com ?

marcusvinicius178, 2021-07-21:
Hi @EricBoiseLGSVL  I am facing the same issue here. I have deleted all the cluster for the new simulator release. However I am not able to create.

I am using 2 e-mails account. Both of them are not able to add clusters

marcusvini178@gmail.com
marcus.posgrad.ufsc@gmail.com

Could you please release the cluster creation for me please? I am in a hurry with my survey deadline. I will be very very grateful if you re-enabale this to me. 

Thanks in advance

eagafonovlge, 2021-07-22:
@krtt 
Could you please check again if you can create new clusters?

krtt, 2021-07-22:
I checked and yes, now I can create new clusters. Thanks for solving this! What was the root cause?

EricBoiseLGSVL, 2021-07-22:
Great! thanks @eagafonovlge .  We have had a few issues with this latest release, sorry about the blocker.

closed 2021-07-22 by EricBoiseLGSVL

marcusvinicius178, 2021-10-01:
Hi @eagafonovlge , @EricBoiseLGSVL unfortunately I have also reached the cluster limit. Could you please release to me too?
The email is marcuslgsvl@gmail.com 

Thanks

thisisdavidfrank, 2021-10-01:
@marcusvinicius178, the limit has been increased to 10 clusters.

marcusvinicius178, 2021-10-01:
@thisisdavidfrank did you increase now? or you mean in general?
I ask excuses, because I use a lot os SVL releases and different machines, so sometimes I add new cluster, unlink the simulation, delete old cluster...etc...
So maybe the system thinks that I am using more than 10 clusters but in fact I am using just one or 2, nut I have deleted the old ones...and maybe the system keep counting they exist....
what do you think about it? Is this the problem?
Or from now on, have I the possibility to add more 10 clusters?