lgsvl/simulator #1582: Random Traffic is not working with self-build simulator

joeking11829, 2021-07-14:
Hi LGSVL Teams,

Reference to #1562
self-build simulator from branch: release-2021.2  ( only modify some tag configuration. )

I found that random traffic is not working, when I use the self-build simulator.

[image]

[image]

Any Idea to fix it ?
Thanks

EricBoiseLGSVL, 2021-07-14:
Did you copy over the npcs and pedestrian asset bundles to the new binary you made?  If you look in the release binary, in the AssetBundles folder, we place the bundles.  Just copy the whole folder.  We hope to support in webui soon.

joeking11829, 2021-07-14:
> Did you copy over the npcs and pedestrian asset bundles to the new binary you made? If you look in the release binary, in the AssetBundles folder, we place the bundles. Just copy the whole folder. We hope to support in webui soon.

Thanks for the reply.
It works fine after I copy the npcs and pedestrian asset bundles to self-build simulator.

BTW, the internet connection of your website (https://wise.svlsimulator.com/) is unstable today.

EricBoiseLGSVL, 2021-07-14:
Glad to help and thanks for letting us know. @thisisdavidfrank

closed 2021-07-15 by joeking11829