lgsvl/simulator #1583: Cannot create a new cluster!!!

wobushigaoying, 2021-07-14:
Hello

I am using the latest version of simulator version 2021.2.

I have unlinked the simulator and deleted the cluster of simulator version 2021.1. 

When I click "Create cluster" button, the website reports an error below:

[image]

FiIipe, 2021-07-14:
I have the same problem. I think they are changing things right now and everything is a bit unstable. The webui is also super slow and I can't delete vehicles.

ebadi, 2021-07-14:
The same problem:
[image]

ebadi, 2021-07-14:
The issue is now resolved for me.

EricBoiseLGSVL, 2021-07-14:
@thisisdavidfrank

wobushigaoying, 2021-07-15:
haha, I had transferred to the old version, 2021.1.  The problem has gone. :)

zelenkovsky, 2021-07-15:
I'm closing this issue, based on the comment above. Please re-open if that happens again.

PS: we've identified the reason for the 504 HTTP error. It is temporarily patched. A proper fix is yet to come. Would you mind reporting if you see 504 HTTP errors again?

DZiYou, 2021-07-16:
I got the same issue, cannot create a new cluster.
Is it possible to launch an offline version?

ihsanyalcinkaya, 2021-07-16:
Same here. I cannot create a new cluster or start a new simulation.

FiIipe, 2021-07-16:
I have the same problem. Looks like it happens from time to time.

thisisdavidfrank, 2021-07-16:
@wobushigaoying, @FiIipe, @ebadi, @DZiYou , @ihsanyalcinkaya: Sorry about this. We've found the cause of this, and we are actively working on fixing the issue. Thank you for bringing this up.

thisisdavidfrank, 2021-07-19:
@wobushigaoying, @FiIipe, @ebadi, @DZiYou , @ihsanyalcinkaya: The issue has been resolved and the fix has been deployed. Please let us know if you see this problem happening again. Thank you.

closed 2021-07-19 by thisisdavidfrank