lgsvl/simulator #1584: How to create 'development' versioned environment and vehicle?

Maestroxr, 2021-07-14:
Hey,
in version 2021.1 all my assets were automatically versioned 'development' when I  built them.
But now I can not build a custom map and have it versioned 'development', instead it gets built as 2021.2
Changing the version from the developer options didn't help.
Also uploading the asset to the preview version of the WebUI doesn't work.

The documentation doesn't seem to mention the subject, so any help would be appreciated.

EricBoiseLGSVL, 2021-07-14:
What version/branch of simulator code are you creating content with? What cloud url are you connecting to?

Maestroxr, 2021-07-18:
Branch: 2021.2 release
Cloud url: tried both wise.svlsimulator.com and wise.preview.svlsimulator.com

EricBoiseLGSVL, 2021-07-19:
Please use wise.svlsimulator.com for release-2021.2.
Use wise.preview.svlsimulator.com for master.
Do not mix the releases with master/preview because it will cause compatibility issues.  Master is only to see features in development and is not a stable version to work on.  I recommend wiping appdata and relinking to release 2021.2 first.  run a sim with editor and assets provided, then build your map once this is working.

What do you mean "Changing the version from the developer options didn't help"? What did you change?

nparkstar, 2021-07-21:
I have the same issue.
Assets built with version 2021.2 have '2021.2' in 'Compatible Simulator Versions' of Asset Bundle information on SVL Simulator Web.
The simulations that contains these assets can't select cluster running in Unity Editor because the assets's version don't have 'development'.
Should I build assets with SVL Simulator 2021.1.1 to run them in Unity Editor?

EricBoiseLGSVL, 2021-07-21:
We are going to release a fix for this very soon