lgsvl/simulator #1585: Cannot upload a custom vehicle asset built with 2021.2

Maestroxr, 2021-07-14:
Hey,
I've been trying to upload a vehicle which worked in previous versions.
I rebuilt the asset, but every time I try to upload it I get an error.
I tried uploading to the preview version WebUI as well as following the documentation on how to build a vehicle.
I get the following error:

[image]

FiIipe, 2021-07-14:
Hi, 

I also faced the same issues, I don't think the 2021.2 is stable. You should use the following versions: 
Unity: 2019.4.18f1
Simulator: 2021.1.1 (https://github.com/lgsvl/simulator/tree/2021.1.1)

Best, 
Filipe

Maestroxr, 2021-07-14:
OK, will do, thanks.

Maestroxr, 2021-07-14:
So far, I was unable to upload a vehicle also built with 2021.1 version, I get the same error.
Same with the preview version.

EricBoiseLGSVL, 2021-07-14:
@eagafonovlge @thisisdavidfrank

DZiYou, 2021-07-15:
I've had the same problem, not just vehicle resources. Maps and bridges. 
Frequently report "invalid resources" and other errors. 
But the same compressed file is not changed again. After waiting for more than ten minutes, it can be uploaded.

wobushigaoying, 2021-07-15:
The same error was reported when I tried to upload a new plugin. 

Unity: 2019.4.18f1
Simulator: 2021.1

Maestroxr, 2021-07-21:
I can confirm this happens as well when using previous version:
- Unity version 2019.4.18f1
- Simulator 2021.1

EricBoiseLGSVL, 2021-07-23:
We are creating a new patch release for 2021.2 and back porting it to 2021.1, this should release very soon.  Sorry about the issue.