lgsvl/simulator #1586: How to use ros2 to output lane lines

DZiYou, 2021-07-14:
Hello, 
I want to use ros2 to output lane information. I noticed that there is lane-line sensor, but it only supports cyberrt, and there is no relevant source code. I would like to ask, is it OK to create the corresponding ros2 message? Can I reuse this sensor, or do I need to write my own sensor?

thank you

EricBoiseLGSVL, 2021-07-14:
I feel we should open source this sensor since many users need lane data.  What do you think @hadiTab @lemketron ?

DZiYou, 2021-07-16:
Will there be an open source plan in the near future? 
If I'm going to build my own sensor, how can I get the opendrive file or map annotation information in the new sensor script? And the XY coordinates and attitude of the car
thankyou

EricBoiseLGSVL, 2021-07-16:
We'll discuss internally about open sourcing the sensor but in the meantime you can look at MapManager.cs and our export tools to see how we parse the data.  Again, the annotation in simulator is not hdmap data.  It is a subset of all hdmap types, plus NPC navigation data.  The Map tool is where the data is converted to the format and this is not done at runtime, only on export and not retained.  LanLineSensor reads the annotation data in the scene and makes meshes that the sensor parses.  Even open source it might not be what you want.

MapOrigin and the transform has the positional data.

lemketron, 2021-07-16:
Actually, the changes needed should be limited to the ROS2 bridge plug-in component so you should not need the LaneLineSensor (though we should still open that code to show people how it works and how to create a virtual sensor like that).

@kamkolak-lge provided this explanation:
> Sensors (including LaneLineSensor) output Simulator-specific formats that don't match neither CyberRT nor ROS. Each bridge then defines implementation that converts this to its own specific format. Currently the only bridge that has an implementation for LaneLinesData (present in Simulator) is CyberRT. ROS2 would require a conversion method between LaneLinesData and ROS-compatible line data. LaneLineSensor shouldn't need any changes.

Check out the [CyberRT bridge project](https://github.com/lgsvl/CyberRT) and you can use that information to add this functionality to the [ROS2 bridge project](https://github.com/lgsvl/ROS2).

DZiYou, 2021-07-19:
Thank you for your advice,
I checked cyberconversions.cs, but in the
`public static apollo.perception.PerceptionLanes ConvertFrom(LaneLinesData data)`
I only found the type of lane in the car. I also need curvature information for lane lines, similar to
`static apollo.perception.camera.LaneLineCubicCurve Convert(LaneLineCubicCurve c)`
But I don't know how to use it. 
For example, there are lane lines on the left and lane lines on the right. How can I extract the corresponding lane line equation and assign it to my message?

kamkolak-lge, 2021-07-19:
@DZiYou `LaneLinesData` is a container with data about multiple lane lines. It contains a list of `LaneLineData`, each of which describes a single line (type, position and shape). Each `LaneLineData` contains struct `LaneLineCubicCurve`, which describes curvature of the line using third-degree polynomial. All of the mentioned types can be found in [this file](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Data/LaneLineData.cs).

How the polynomial coefficients should be interpreted is described in the [sensor documentation](https://www.svlsimulator.com/docs/simulation-content/lane-line-sensor/#curve-definition).

To add support for `LaneLinesData` to your bridge of choice, you'd have to register the publisher that can perform the conversion. `IBridgeFactory` interface exposes `Register(IBridgePlugin plugin)` method that registers all publishers and subscribers for the bridge plugin. In case of CyberRT, you can see that there's a converter specified for `LaneLinesData`:
```
RegPublisher<LaneLinesData, apollo.perception.PerceptionLanes>(plugin, CyberConversions.ConvertFrom);
```
which points to the `ConvertFrom` method you mentioned. A simplified way of looking at what this line does would be: "if any sensor wants to send `LaneLinesData` type data (Simulator-specific, bridge independent), convert it to `apollo.perception.PerceptionLanes` (Apollo-specific type) using the specified `CyberConversions.ConvertFrom` method.

Adding this to any other bridge would be very similar - just register the appropriate publisher with matching target type (specific for the bridge in question) and matching converter.

DZiYou, 2021-07-19:
> @DZiYou `LaneLinesData` is a container with data about multiple lane lines. It contains a list of `LaneLineData`, each of which describes a single line (type, position and shape). Each [...]

@kamkolak-lge  thank you
`RegPublisher<LaneLinesData, apollo.perception.PerceptionLanes>(plugin, CyberConversions.ConvertFrom);`
points to the `ConvertFrom` . But in CyberConversions.cs
the `public static apollo.perception.PerceptionLanes ConvertFrom(LaneLinesData data)`  only includ the Line type and position .not include shape.
the line type include in `static apollo.perception.camera.LaneLineCubicCurve Convert(LaneLineCubicCurve c)`
where or which trigger the `static apollo.perception.camera.LaneLineCubicCurve Convert(LaneLineCubicCurve c)`.
and what's the difference between `Convert` and `ConvertFrom`

kamkolak-lge, 2021-07-23:
Conversion loop for lines data processes each line curvature in [object initializer](https://github.com/lgsvl/CyberRT/blob/777778574e7d01d20979572288dbbe787797c4e9/CyberConversions.cs#L520):
```
var line = new apollo.perception.camera.CameraLaneLine()
{
    curve_camera_coord = Convert(lineData.CurveCameraCoord)
};
```
and then [adds it](https://github.com/lgsvl/CyberRT/blob/777778574e7d01d20979572288dbbe787797c4e9/CyberConversions.cs#L583) to `camera_laneline` in the `result`:
```
result.camera_laneline.Add(line);
```
Curvature for each line should be present in `camera_laneline` array, as shown here:
[image]

Don't pay too much mind to the difference in `Convert` and `ConvertFrom` - it's just done this way so all publicly available methods from this class specify whether they convert to or from given format. Non-public members just don't always follow this convention.

DZiYou, 2021-07-24:
@kamkolak-lge Thankyou ，Thank you. I have achieved the desired function

closed 2021-07-26 by EricBoiseLGSVL