lgsvl/simulator #1587: Offline mode simulation craches

krtt, 2021-07-14:
When running the sim in the offline mode (first it runs ok), then stopping it, and then starting the sim again (in offline mode), the program crashes and does not run or close any more. Only killing the process helps. The sim I tried several times was 'API Only'.

Ubuntu 18.04
Sim version 2021.1.1
player.log attached
[Player.log](https://github.com/lgsvl/simulator/files/6817328/Player.log)

EricBoiseLGSVL, 2021-07-14:
Playerlog doesn't have any crash info, odd.  We'll look into this, thanks for posting. @revati-naik let's file a bug and see if this is happening in 2021.2

EricBoiseLGSVL, 2021-07-26:
I recommend moving to 2021.2.1 because I cannot reproduce this issue.  I ran 04 quick script 10 times in offline mode API and no crashes.  Are you running a particular script that you can share?

styj5, 2022-03-21:
Hi,

is it possible to transfer Assets in the old account in old LGSVL old release to 2021.2.1?

styj5, 2022-03-21:
@EricBoiseLGSVL  Thanks in advance

EricBoiseLGSVL, 2022-03-21:
Sure, be aware that the current simulator backend will end support soon.  If you do this you will need to open the assets in the new simulator, update the shaders to hdrp supported shaders and then build the new updated assets.

styj5, 2022-03-22:
kk, thanks a lot^^