lgsvl/simulator #1593: Problem with odom to base_link transfrom

rythm07, 2021-07-15:
Hello,

Apparently, the simulator doesn't publish itself the odom->base_link transform over ROS (without autoware integration). 

Is there any way that I can't see to enable it or do I just have to make my own broadcaster to publish that transform?

Thanks for your help!

lemketron, 2021-07-15:
@hadiTab ?

rythm07, 2021-07-19:
Does anyone have an answer for me?

sisaha9, 2021-07-19:
@rythm07 You need to publish odom -> base_link on your own. You can use robot_localization to publish odom -> base_link and map -> odom if you don't want to use autoware.auto. If you want to just use dummy ones you can use static_transform_publisher to publish these transforms as well

rythm07, 2021-07-19:
I figured it out. I'm using robot_pose_ekf to publish odom->base_link and the slam_toolbox which publishes map->odom.
Thanks for your answer!

closed 2021-07-19 by rythm07