lgsvl/simulator #1594: Unable to Launch the simulator on ubuntu

rajeshjoshi8792, 2021-07-15:
Hi,

i m executing simulator in ubuntu. but its not opening. 

i m using Tightvnc to connect my server.

here is my player.log file.
Mono path[0] = '/home/ubuntu/Rajesh/svlsimulator-linux64-2021.2/simulator_Data/Managed'
Mono config path = '/home/ubuntu/Rajesh/svlsimulator-linux64-2021.2/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1024x768 (primary device).
Loading player data from /home/ubuntu/Rajesh/svlsimulator-linux64-2021.2/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1024 x 768 @ 0 Hz
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Vulkan detection: 0
No supported renderers found, exiting

please help me resolve this issue.

lemketron, 2021-07-15:
Please refer to #1431 and #1503 (or [search issues for "VNC"](https://github.com/lgsvl/simulator/issues?q=is%3Aissue+vnc+))

rajeshjoshi8792, 2021-07-19:
i refer all the topics related to simulator not opening. and tried all the steps. but still simulator not opening.

but this times, the error is different. please check my player.log file

Mono path[0] = '/home/ubuntu/Rajesh/SVL Simulator/svlsimulator-linux64-2021.2/simulator_Data/Managed'
Mono config path = '/home/ubuntu/Rajesh/SVL Simulator/svlsimulator-linux64-2021.2/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1024x768 (primary device).
Loading player data from /home/ubuntu/Rajesh/SVL Simulator/svlsimulator-linux64-2021.2/simulator_Data/data.unity3d
Desktop is 1024 x 768 @ 0 Hz
Xlib:  extension "GLX" missing on display ":1.0".
Xlib:  extension "GLX" missing on display ":1.0".
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2020.3.3f1 (76626098c1c4)
[Subsystems] Discovering subsystems at path /home/ubuntu/Rajesh/SVL Simulator/svlsimulator-linux64-2021.2/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
Xlib:  extension "GLX" missing on display ":1.0".
Xlib:  extension "GLX" missing on display ":1.0".
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil)
[Vulkan init] Physical Device 0x8c9f90 [0]: "Tesla T4" deviceType=2 vendorID=10de deviceID=1eb8
[Vulkan init] Selected physical device 0x8c9f90
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=128
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=1
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[... 179 lines cut ...]

found manifest at Pedestrians/Walkers/pedestrian_Robin/manifest.json

found manifest at Pedestrians/Walkers/pedestrian_Stephen/manifest.json

found manifest at Pedestrians/Walkers/pedestrian_Zoe/manifest.json

Loaded NPCs behaviours: 3  NPC models: 7  Pedestrians: 12 Controllables: 3 Bridges: 0in 00:00:00.8520795

Current Database Version: 8

Final Database Version: 8

Caught fatal signal - signo:11 code:1 errno:0 addr:0x50
Obtained 10 stack frames.
#0  0x007fb723473980 in funlockfile
#1  0x007fb724b57ee7 in PhysicsManager::GetDefaultPhysicsSceneHandle() const
#2  0x007fb724a94ceb in UnityScene::~UnityScene()
#3  0x007fb724a91cac in RuntimeSceneManager::~RuntimeSceneManager()
#4  0x007fb724a91525 in StaticDestroyRuntimeSceneManager(void*)
#5  0x007fb7247d6fff in RegisterRuntimeInitializeAndCleanup::ExecuteCleanup()
#6  0x007fb724a7695a in RuntimeCleanup()
#7  0x007fb724c521f6 in PlayerMain(int, char**)
#8  0x007fb723091bf7 in __libc_start_main
#9  0x00000000201029 in _start

lemketron, 2021-07-19:
> Another instance of simulator is already using this data folder: /home/ubuntu/.config/unity3d/LGElectronics/SVLSimulator

Are you running more than one Simulator instance?

rajeshjoshi8792, 2021-07-20:
No..i tried to open simulator. But its not opening. So i again tried to open. But getting same issue. I got log in player.log file that simulator connecting to wise.svlsimulator.com. but im unable to see simulator in desktop.

lukekenworthy, 2022-08-30:
@rajeshjoshi8792 Did you ever find out how to solve this?