lgsvl/simulator #1595: I cannot play CubeTown on Unity

NobushigeHaraki, 2021-07-16:
I am a beginner. For practice, I downloaded the open source "Cube Town" and opened it in Unity. After that, I clicked the "Play" button to test on Unity, but I got the following error. How can I resolve the error?
I use Ubuntu16.04.

error:
①
NoSuccessException: Simulator version'Development' does not match to the linked version'dev' (400)
CloudAPI + <Connect> d__9.MoveNext () (at Assets / Scripts / Web / ConnectionManager.cs: 420)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <9577ac7a62ef43179789031239ba8798>: 0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <9577ac7a62ef43179789031239ba8798>: 0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <9577ac7a62ef43179789031239ba8798>: 0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <9577ac7a62ef43179789031239ba8798>: 0)
System.Runtime.CompilerServices.TaskAwaiter`1 [TResult] .GetResult () (at <9577ac7a62ef43179789031239ba8798>: 0)
Simulator.Loader + <StartSimulation> d__40.MoveNext () (at Assets / Scripts / Web / Loader.cs: 317)
UnityEngine.Debug: LogException (Exception)
Simulator. <Start Simulation> d__40: MoveNext () (at Assets / Scripts / Web / Loader.cs: 417)
UnityEngine.UnitySynchronizationContext: ExecuteTasks () (at /home/bokken/buildslave/unity/build/Runtime/Export/Scripting/UnitySynchronizationContext.cs:107)

➁
Cannot load scene: Invalid scene name (empty string) and invalid build index -1
UnityEngine.SceneManagement.SceneManager: LoadScene (String)
Simulator. <Start Simulation> d__40: MoveNext () (at Assets / Scripts / Web / Loader.cs: 423)
UnityEngine.UnitySynchronizationContext: ExecuteTasks ()

[image]

EricBoiseLGSVL, 2021-07-16:
You cannot open a scene and then just hit play in the Unity Editor.  You need to go to the LoaderScene in Assets/Scenes/. Start the editor from there.  You will be prompted to link to cloud and then you can run simulations.  We are working on development settings in the Simulator menu that will allow you to run from a scene but it is still WIP and has issues with sensors this release.  Hopefully we will have it fixed soon.

closed 2021-07-21 by NobushigeHaraki

tianyunzhe, 2021-11-20:
> 
When I start LoaderScene.unity and click "LINK TO CLOUD", there is no response.  How can I solve the error? 

The "Development Settings" is as follow.

[image]

Many thanks!

EricBoiseLGSVL, 2021-11-22:
What is your default browser?  Do you have any software blocking access to the web?

EricBoiseLGSVL, 2021-11-22:
Open the wise.svlsimulator.com in the browser first, be signed in, then hit the link to cloud button.  Also, be sure to close the development settings window if you are not linked, nor starting the editor in a scene, not loaderscene.  That is for scene testing.