lgsvl/simulator #1596: API Only sim with Apollo sometimes does not work

krtt, 2021-07-16:
I created a simple test scenario script with the default Lincoln2017MKZ ego car and our own custom map. It run fine with Apollo several times. Then, without any changes it sometimes did not run correctly (the ego car did not start driving sometimes). And then, without any changes, it stopped working at all (ego did not start driving at all). I found seemingly 2 problems. 

First problem. When it sometimes did not run, it appeared to be because of Apollo's planning module. The planning module looks enabled but it gets somehow stuck and even while showing that it is enabled, it is actually not working properly. The only workaround for this was to turn the Planning module in Apollo Dreamview OFF and back ON. Then the car immediately started driving. Other attempts to restart Planning module did not work ([disabling it via api command](https://www.svlsimulator.com/docs/python-api/dreamview-api/#disable_module) and [enabling it again via api command](https://www.svlsimulator.com/docs/python-api/dreamview-api/#enable_module)). Player log: 
[Player_15.07.2021_apollo_planning_module.log](https://github.com/lgsvl/simulator/files/6830722/Player_15.07.2021_apollo_planning_module.log)

Second problem. When it stopped working at all, it appeared to be because of all Apollo's modules. None of the modules are enabled any more eventhough I have not changed anything in how I run Apollo, how I run the sim, any assets, and anything in my python script. Player.log: 
[Player_16.07.2021_does_not_work_after_no_changes_made.log](https://github.com/lgsvl/simulator/files/6830731/Player_16.07.2021_does_not_work_after_no_changes_made.log)

How can I fix this?

Ubuntu 18.04
Apollo6 (I use modular testing only)
Sim v2021.1.1
Python script [test_0.zip](https://github.com/lgsvl/simulator/files/6830745/test_0.zip)
attached

krtt, 2021-07-16:
And now, without any changes done, it suddenly runs ok - the car starts driving again. 
[Player.log](https://github.com/lgsvl/simulator/files/6831088/Player.log)

Can you help me find the root cause?

lemketron, 2021-07-16:
In time working with Apollo you will likely get to know the various reasons why certain modules might block the vehicle from moving. We've discussed some of them in [other simulator issues](https://github.com/lgsvl/simulator/issues?q=is%3Aissue+apollo+planning+routing+drive). In general the simulator is usually working ok but there is some reason why Apollo is stuck. 

You'll need to learn how to read the Dreamview screen to understand what's happening and what is blocking Apollo from driving. You can read the [Dreamview documentation](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/dreamview_usage_table.md

) for more information: 

If you have further issues with Apollo please ask on the [Apollo issues page](https://github.com/ApolloAuto/apollo/issues).