lgsvl/simulator #1597: lgsvl gps data suspect

asobhy-qnx, 2021-07-16:
I am experiencing localization issues when using gps on LGSVL with autoware.auto.

I replaced ndt localization with gps and imu positioning. I am using this map asset: https://assets.lgsvlsimulator.com/ec057870762b5a967a451c93444b67d0b64e9656/environment_AutonomouStuff

First thing I am seeing is that the altitude the gps is recording is -18.9m which does not seem right. I do not believe AutonomousStuff parking lot is 18.9 meters below surface of the ocean.

Second thing is that when I place the vehicle and on the origin on both the lanelet2 map and LGSVL map the two maps are aligned perfectly, then when I start moving the vehicle in the East direction I see the car position change in the East-South direction on the lanelet2 map.

I went over my calculations so many times and I cannot find any issues which makes me suspect that the gps coordinates might not be mapped correctly on LGSVL.

My calculations to get the position of the vehicle on the map are as following:

1- Get base_link position from Geodetic (lat,lon,alt) measurements from GPS /gnss/fix topic.
2- Convert Geodetic measurments to Geocentric (x,y,z).
3- The Geocentric measurments calculated are for the base_link on earth which is equivalent to the translation from earth_to_base_link.

EricBoiseLGSVL, 2021-07-16:
Current version of autonomoustuff is rotated correctly.  Thanks for the help on the offset, next release we will have it corrected to 14m.

asobhy-qnx, 2021-07-16:
How can I make sure I have the current version of the map?

EricBoiseLGSVL, 2021-07-16:
release 2021.2 tag.  2021.2 map will not work with 2020.06 simulator.

asobhy-qnx, 2021-07-19:
@EricBoiseLGSVL I tested using 2021.2 and the behaviour in 2020.06 is the same as in 2021.2.

Tested on LGSVL 2020.06 & 2021.02
The problem:
When driving on the map in East or West direction Lat value for GPS increase or decrease respectively. This indicate a misalignment in the gps data on LGSVL map.

How to reproduce:

1- Manually drive the vehicle until you get an orientation on the IMU equal to [0,0,0,1], heading of zero degrees, which puts the vehicle into a heading in the East direction, as confirmed with google maps. Don't know why the IMU has it's zero heading facing east, if it's frame of reference is matching with base_link, where x is front and y is left.

2- Once the vehicle is facing east, drive the car in straight line and observe the value for the latitude on LGSVL GPS reading, you will see that the value is changing. Shouldn't the Latitude be fixed if we are driving at a zero angle East?

EricBoiseLGSVL, 2021-07-19:
Hmm, @hadiTab it this expected behavior? I checked the map and it is rotated correctly.  Would the MapOrigin be wrong?  We'll look into it @asobhy-qnx

asobhy-qnx, 2021-07-26:
@EricBoiseLGSVL it seems that it was an incorrect calculation from my end. The data in LGSVL is good.

closed 2021-07-26 by asobhy-qnx